node-hid/node-hid #353: NODE_MODULES_VERSION still wrong after rebuild?

Newbrict, 2019-11-25:
I followed the instructions from this section https://github.com/node-hid/node-hid#electron-projects-using-node-hid

I still get the error `NODE_MODULE_VERSION` error:
[image]

Here's the output from my `npm install`:

```
$ npm install

> node-hid@1.0.0 install C:\Users\newbr\Desktop\Code\atompalm\electron\manager\node_modules\node-hid
> prebuild-install || node-gyp rebuild

> core-js@3.4.2 postinstall C:\Users\newbr\Desktop\Code\atompalm\electron\manager\node_modules\core-js
> node -e "try{require('./postinstall')}catch(e){}"

Thank you for using core-js ( https://github.com/zloirock/core-js ) for polyfilling JavaScript standard library!

The project needs your help! Please consider supporting of core-js on Open Collective or Patreon:
> https://opencollective.com/core-js
> https://www.patreon.com/zloirock

Also, the author of core-js ( https://github.com/zloirock ) is looking for a good job -)

> electron@7.1.2 postinstall C:\Users\newbr\Desktop\Code\atompalm\electron\manager\node_modules\electron
> node install.js

> manager@1.0.0 postinstall C:\Users\newbr\Desktop\Code\atompalm\electron\manager
> electron-rebuild --force

- Searching dependency tree
√ Rebuild Complete
npm notice created a lockfile as package-lock.json. You should commit this file.
npm WARN manager@1.0.0 No repository field.

added 258 packages from 216 contributors and audited 600 packages in 7.958s

2 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
```

and as can be seen in my `package.json` I put the rebuild steps in:
```json
$ cat package.json
{
  "name": "manager",
  "version": "1.0.0",
  "description": "n/a",
  "main": "main.js",
  "scripts": {
    "start": "electron .",
    "postinstall": "electron-rebuild --force"
  },
  "author": "Dimitar Dimitrov",
  "license": "ISC",
  "devDependencies": {
    "electron": "^7.1.2",
    "electron-rebuild": "^1.8.8",
    "node-gyp": "^6.0.1",
    "node-hid": "^1.0.0"
  }
}
```

any tips would be greatly appreciated, thank you

todbot, 2019-11-25:
Hi @Newbrict,
Does electron-hid-toy work for you?  https://github.com/todbot/electron-hid-toy

Newbrict, 2019-11-25:
> 
> 
> Hi @Newbrict,
> Does electron-hid-toy work for you? https://github.com/todbot/electron-hid-toy

Yes I just cloned, `npm install` & `npm start` and it worked

Newbrict, 2019-11-25:
I fixed the issue:
It was caused by `node-hid` being in `devDependencies` instead of `dependencies`.

Working setup:
```
  "dependencies": {
    "node-hid": "^1.0.0"
  },
  "devDependencies": {
    "electron": "^7.1.2",
    "electron-rebuild": "^1.8.8"
  }
```

closed 2019-11-25 by Newbrict