node-hid/node-hid #354: handle.on("data") read breaks electron / chrome

Newbrict, 2019-12-07:
Hi,

I have a piece of code like this:
```js
    handle = new hid.HID(device.path);
    console.log(handle);
    handle.on("data", (data) => {
      console.log("data came in");
    });
```
the `console.log(handle)` is a valid handle which I can use to write to the device without issue. When I add the `handle.on("data"...` handler immediately after I see no errors either in my console or from the terminal that I launched electron. the screen goes white and devtools says "DevTools was disconnected from the page. Once page is reloaded, DevTools will automatically reconnect."

Screenshot:

[image]

since there are no errors I am not sure how to even begin debugging this.. Please advise. Thank you.

todbot, 2019-12-07:
Hi,  Yes, that is really strange.
A few questions/comments:
1. What OS, node-hid version, and Electron version are you using?
2. Is this code running in the main process or in the renderer process?  If in the renderer, it should be in the main process. (I don't think the invisible Electron remote-ing works well with native modules)
3. Can you create a small command-line Node.js program that has the same basic communication structure as the Electron code?  
4. This shouldn't matter, but does `device.on('data', function(data) { console.log(data) });` produce different results?
5. Just curious, what device is this?

Newbrict, 2019-12-07:
1. Windows 10 Pro (1803, build 17134.1069).  "node-hid": "^1.0.0", "electron": "^7.1.2",

2. it is running on the render process (although the writes worked perfectly in the render process), I moved the code to the main process and now I get an error:
[image]
nothing in the main process' console though

3. https://github.com/Newbrict/node-hid-broken-example

4. Same result

5. Can't say, although it's not a mouse or a keyboard if you're concerned about the windows permission.

Thanks for the quick response!

todbot, 2019-12-07:
When you say you can't say, you mean you do not know or because it's proprietary?

Newbrict, 2019-12-07:
The device is proprietary

todbot, 2019-12-08:
Hi,
I just tried your test application (thank you btw) using a Teensy 2.0 configured as a [RawHID device](https://www.pjrc.com/teensy/rawhid.html) (using the [basic Teensyduino RawHID example](https://github.com/PaulStoffregen/Teensyduino_Examples/blob/master/USB_RawHID/Basic/Basic.pde)), and the application connects to the device and correctly receives data from it.  This works on both MacOS 10.14.6 and Windows 10 Pro 1903. (Oh and in your `main.js` you do not define `update_status_label()` but I just replaced that with a `console.log()`)

Have you verified your device works via command-line tools or non-Node.js-based means?  Often if the HID descriptor or config descriptor is defined in a way that Windows doesn't like, the HID subsystem gets cranky. 

Two examples of this crankiness I'm intimately familiar with: 
1. Multiple usages defined in the HID descriptor must have Report Ids defined and be unique. Otherwise Windows HID ignores the device, but you can see in Device Manager your device has a yellow badge with a message like:  "The parser discovered a top level collection in a complex device (more than one top level collection) that had no declared report ID or a report ID spanned multiple collections."
2. Feature reports must be exactly the size in bytes that's defined in the HID descriptor or Windows ignores them, while Linux & MacOS will work.

You may find the [hidapitester command-line tool](https://github.com/todbot/hidapitester) useful. It's meant to be a simple exerciser of the `hidapi` library but can be useful in sending/receiving data to HID devices.

Newbrict, 2019-12-08:
Thank you very much for all the tips, I was able to make everything work. It turns out my device was misconfigured, details are as follows:

report descriptor was
```
    0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
    0x09, 0x01,        // Usage (0x01)
    0xA1, 0x01,        // Collection (Application)
    0x09, 0x02,        //   Usage (0x02)
    0x15, 0x00,        //   Logical Minimum (0)
    0x26, 0xFF, 0x00,  //   Logical Maximum (255)
    0x75, 0x08,        //   Report Size (8)
    0x95, 0x40,        //   Report Count (64)
    0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x09, 0x03,        //   Usage (0x03)
    0x15, 0x00,        //   Logical Minimum (0)
    0x26, 0xFF, 0x00,  //   Logical Maximum (255)
    0x75, 0x08,        //   Report Size (8)
    0x95, 0x40,        //   Report Count (64)
    0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0xC0,              // End Collection
```
which does not specify an input (device -> host)
so I had to change it to the following, once flashed everything works perfectly (including the hidapitester, great tool thanks for showing me / creating it)
```
    0x06, 0x00, 0xFF,  // Usage Page (Vendor Defined 0xFF00)
    0x09, 0x01,        // Usage (0x01)
    0xA1, 0x01,        // Collection (Application)
    0x09, 0x02,        //   Usage (0x02)
    0x15, 0x00,        //   Logical Minimum (0)
    0x26, 0xFF, 0x00,  //   Logical Maximum (255)
    0x75, 0x08,        //   Report Size (8)
    0x95, 0x40,        //   Report Count (64)
    0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0x09, 0x03,        //   Usage (0x03)
    0x15, 0x00,        //   Logical Minimum (0)
    0x26, 0xFF, 0x00,  //   Logical Maximum (255)
    0x75, 0x08,        //   Report Size (8)
    0x95, 0x40,        //   Report Count (64)
    0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
    0xC0,              // End Collection
```

Thanks again for all the help!!

closed 2019-12-08 by Newbrict

Newbrict, 2019-12-08:
also I believe what was happening to cause the crash would be fixed by wrapping the .on("data") in a try/catch

I haven't tested that but that's my suspicion. I guess the issue on my end was caused by windows not binding the the uib socket since the descriptor didn't specify it as an input.

todbot, 2019-12-08:
Interesting! Thanks for the follow-up.  This is definitely an error condition I would've have expected.