node-hid/node-hid #355: Question: Is this working on new circleci windows executor?

ty-dev, 2019-12-07:


todbot, 2019-12-07:
I've no clue what that is. Could you reference an existing example?

ty-dev, 2019-12-09:
CircleCI Windows Executor: https://circleci.com/docs/2.0/hello-world-windows/

Appears it was a electron + node-hid on windows combination when using electron-rebuild. Closing ticket but if it helps anyone in the future, need to  use vs2105 environment to get things working:
```
# .circleci/config.yml
version: 2.1

orbs:
  win: circleci/windows@2.2.0

jobs:
  build:
    executor: win/default
    steps:
      - checkout
      - run: nvm install 8.10.0
      - run: nvm use 8.10.0
      - run: node -v
      - run: npm install --global --production windows-build-tools --vs2015
      - run: npm install node-gyp
      - run: npm install node-gyp -g
      # other build steps

```

closed 2019-12-09 by ty-dev