node-hid/node-hid #359: Cannot build from source

danbulant, 2019-12-30:
Windows_NT 10.0.19025 (Slow ring Insider)

Tried using node-hid (required as dependency from dualshock-controller), as prebuilt wasn't made, needed to build from source.
Error:
```
..\src\HID.cc(207): error C2661: 'v8::Function::NewInstance': no overloaded function takes 2 arguments [PATH TO PROJECT\node_modules\node-hid\build\HID.vcxproj]
```

Maybe something deprecated that got removed in newer windows?

todbot, 2019-12-30:
Hi,
A few questions:
- What version of Node.js are you using?  
- Is your project using Electron or Nw.js?
- Do you have an example project on github I can try on my Windows system?
- What is "Windows_NT 10"? When I do "VER" in CMD.EXE on my Windows 10 Pro system is reports "Microsoft Windows [Version 10.0.18362.418]"

We offer prebuilds for Windows 8-10, 32-bit & 64-bit, so it's odd you had to compile from source.

danbulant, 2019-12-30:
Windows_NT showed up in the logs. Also, all windows (starting with 2000) are NT versions (last one that wasn't NT is ME). Check it yourself - your browser will say it's Windows NT 10.

Your version is older than mine because (as I already stated) I am in insider preview (because of features such as WSL 2). That might also be why I don't have prebuild binary.

My project is using Electron. 

Node v11.9.0
npm 6.13.1

I just created a bare project and run `npm i dualshock-controller` which has node-hid as dependency. Trying installation of just node-hid didn't work as well.

todbot, 2019-12-30:
Ah yes, the ghost of NT.  I thought maybe you were reporting the output of "ver.exe" or "systeminfo.exe" so was a bit confused. 

What version of Electron?   Are you seeing the error when you do the initial `npm install` or when you do `electron-rebuild`?

Normally the prebuilds are only concerned with Node version as Windows binaries are very compatible across Windows versions.  I'm thinking perhaps you're on a version of Electron that's currently not supported by prebuild (and has yet again deprecated v8 APIs that were okay before)

danbulant, 2019-12-30:
Electron version 1.7.2 (I think it's latest, just did npm install electron)

todbot, 2019-12-30:
Perhaps you mean Electron 7.1.2?
Could you try doing `npm install` on this project? https://github.com/todbot/electron-hid-toy
This project uses Electron 7.1.2 and installs using most any version of Node (I used Node v12.1.0) on Windows 10 Pro x64 version 1903. (It works on other versions of Windows, but that's the version I just now verified on)

It sounds like something is amiss in your Node setup?

I'm not that familiar with Windows dev stuff and don't know what "Slow ring Insider" means. Is there a way for me to easily get one of these preview builds to install on a VM to test against?

danbulant, 2019-12-30:
Oh sorry wrong, electron -v shows v5.0.7

danbulant, 2019-12-30:
Also for the windows isnider, if you go to settings > Update & Security, the last item is Windows Insider Program. Then you can choose which preview you want (slow or fast, slow is recommended)

danbulant, 2019-12-31:
Oh just found out the module uses older (0.5.7) version. Sorry.

closed 2019-12-31 by danbulant