node-hid/node-hid #360: Device don't have on function

danbulant, 2019-12-31:
I tried using node-hid, but for some reason, even if the device is found and have most of functions, the .on is missing.

[image]

todbot, 2019-12-31:
What is the code you used?

danbulant, 2019-12-31:
```js
var devices = HID.devices();
        
        const deviceMeta = devices
            .filter(isController)[0];
        if (deviceMeta) {
            device = new HID.HID(deviceMeta.path);
            console.log(device);
        }
```

todbot, 2019-12-31:
It looks like you're not checking if `device` actually exists before calling `device.on()`

danbulant, 2019-12-31:
Changed the code but it still doesn't work
```js
var devices = HID.devices();
        
        const deviceMeta = devices
            .filter(isController)[0];
        if (deviceMeta) {
            device = new HID.HID(deviceMeta.path);
            console.log(device);
            device.on('data', processFrame.bind(this));
            device.on('error', handleException.bind(this));
        }
```

[image]

todbot, 2019-12-31:
I'm not sure what the issue is. I've never seen this particular problem.

If this is keyboard, mouse, or other device the OS has a device driver for, it could be conflicting and not letting `node-hid` have access.  Or if the device has multiple usages, you may need to specify the exact usage you want to open.

danbulant, 2019-12-31:
It's a Dualshock 4 controller connected over USB.
The problem is that for other people, it works (using dualshock-controller library).

OS doesn't look like having driver for that  (why would Microsoft support Sony?)

At least it looks like that, because it (as I said) works [here](https://github.com/rdepena/node-dualshock-controller)

todbot, 2019-12-31:
Ah yes, I've used that library myself.  In fact, there's a `test-ps3.js` example in the `node-hid/src` showing a simplified way of getting data from a DS3 controller.

I believe Windows has a driver for DS4 controllers now (so you can seamlessly use it in games an such). You should be able to discover this by going into Device Manager and finding the DS4 controller in the device list.  If it shows up as "Game controller" and not just "HID compliant device" then I think `node-hid` cannot access it.

danbulant, 2020-01-01:
oh OK.Just thought it didn't have as others didn't see to have that issue... Will check the drivers tomorrow

Sent via Migadu.com, world's easiest email hosting>

danbulant, 2020-01-01:
The DS4 controller shows up as HID-compliant game controller. Is that a problem?