node-hid/node-hid #361: Using --inspect makes HID no longer working

gre, 2020-01-28:
When you use `node --inspect` , as soon as you start the Profiler, it will make any HID attempt fail with `cannot open device with path ...`.

Is there any way to fix this? It would be pretty handy to be able to debug an app and profile it even when it's using HID.

Thanks

todbot, 2020-01-28:
What OS and node/Electron version are you using?

gre, 2020-01-28:
I was initially testing in context of Electron 7 (with that "cannot open device")

But to focus more at a minimal level, i'm not testing directly on the Terminal using `--inspect-brk` with

```
Node 12.14.1
MacOS Catalina 10.15.2
```

and it seems I'm now getting `Cannot write to HID device` which happens as soon as I do the first `device.write(data)`

(without --inspect-brk or --inspect that have started recording, it seems all is properly working)

todbot, 2020-01-28:
I suspect this may be due to how Catalina handles permissions now.  Are you signing your app and giving it the correct entitlements?
Alternatively, can you test on a non-Catalina Mac?
Also, do you have any other native modules in  your codebase?  I suspect this isn't a problem with node-hid specifically.

gre, 2020-01-28:
Ok thanks!
I'll make a minimal script to see if it still happens and ask more dev in my team to test in different environment.