node-hid/node-hid #363: device.write() returning a number not agreeing with parameter.

ishmal, 2020-02-16:
- Environment:
    Windows 10 Powershell
- Version:
     latest
- Code:
     const nrBytes = device.write([1,2,3,4,5]);
     console.log("nrBytes: " + nrBytes);
- Expected result:
     nrBytes: 6
- Experienced:
     42
 

Am I expecting the wrong result?  According to the docs,  it should be n + 1, right?
Yet I am seeing 42 regardless if there are 1 through 10 array members.
Is there some wrapper payload being sent?  Or is HID.cc returning the wrong thing?
Maybe it just isn't clear to me from the README.

Thanks much!

todbot, 2020-02-16:
Hi,
The number of bytes written depends on the REPORT_COUNT value of the HID report you're sending.   Usually REPORT_COUNT is 64 or 8 or even 1, 42 is odd.

What is the device you're sending to?

ishmal, 2020-02-16:
@todbot    I just debugged around the hid_write call, and the length going in is 11, and the return is 42.   Since this is not a node-hid issue, I'd like to close this.   Thanks for the info.   I suppose I just need to study HID more.   Is there a way to get the actual number of bytes written?
This is a generic Freescale MCU Mouse Demo (default name until you change it) chip.

closed 2020-02-16 by ishmal

todbot, 2020-02-16:
You need to know the HID report descriptor that the device is sending to the PC.  This describes the types of reports, the formats, and their sizes.  On Windows I've never found a good (free) utility to show you HID report descriptors, which is weird considering how strict Windows is with the buffer sizes needing to match the REPORT_COUNT size.

One HID report descriptor resource that may be useful is: 
https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/