node-hid/node-hid #364: node-hid install fails: EACCES: permission denied, access '/root/.npm'

joearkay, 2020-02-17:
Hi, 

Trying to install node-hid on a fresh Raspberry Pi running Raspbian Buster. Stack trace below: 

```
pi@raspberrypi:~ $ sudo npm install -g node-hid
/usr/bin/hid-showdevices -> /usr/lib/node_modules/node-hid/src/show-devices.js

> node-hid@1.2.0 install /usr/lib/node_modules/node-hid
> prebuild-install || node-gyp rebuild

prebuild-install WARN install EACCES: permission denied, access '/root/.npm'
gyp WARN EACCES current user ("nobody") does not have permission to access the dev dir "/root/.cache/node-gyp/12.16.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/tmp/.node-gyp"
gyp ERR! configure error 
gyp ERR! stack Error: EACCES: permission denied, mkdir '/usr/lib/node_modules/node-hid/build'
gyp ERR! System Linux 4.19.97-v7+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/node-hid
gyp ERR! node -v v12.16.0
gyp ERR! node-gyp -v v5.0.7
gyp ERR! not ok 
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-hid@1.2.0 install: `prebuild-install || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the node-hid@1.2.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-17T18_52_55_179Z-debug.log

```

todbot, 2020-02-17:
Hi,
In general you should not use `sudo` with `npm`. See: https://medium.com/@ExplosionPills/dont-use-sudo-with-npm-still-66e609f5f92

closed 2020-02-17 by todbot

joearkay, 2020-02-17:
Hi @todbot - thanks. I've tried running without and I still have issues installing. It seems to be done to a permissions issue? May you please re-open the issue if possible?
```

pi@raspberrypi:~ $ npm install -g node-hid
npm WARN checkPermissions Missing write access to /usr/lib/node_modules
npm ERR! code EACCES
npm ERR! syscall access
npm ERR! path /usr/lib/node_modules
npm ERR! errno -13
npm ERR! Error: EACCES: permission denied, access '/usr/lib/node_modules'
npm ERR!  [Error: EACCES: permission denied, access '/usr/lib/node_modules'] {
npm ERR!   stack: "Error: EACCES: permission denied, access '/usr/lib/node_modules'",
npm ERR!   errno: -13,
npm ERR!   code: 'EACCES',
npm ERR!   syscall: 'access',
npm ERR!   path: '/usr/lib/node_modules'
npm ERR! }
npm ERR! 
npm ERR! The operation was rejected by your operating system.
npm ERR! It is likely you do not have the permissions to access this file as the current user
npm ERR! 
npm ERR! If you believe this might be a permissions issue, please double-check the
npm ERR! permissions of the file and its containing directories, or try running
npm ERR! the command again as root/Administrator.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-02-17T19_12_11_316Z-debug.log
```

todbot, 2020-02-17:
Once you've run `sudo` with `npm` it can screw up your entire node install.  The only solution that's definitively worked for me is to uninstall everything node-releated and re-install.  The problem isn't `node-hid` specific, but any module that has more than just a few .js files.

npmjs.com has some hints on how to fix it: https://docs.npmjs.com/resolving-eacces-permissions-errors-when-installing-packages-globally
which is very similar recommendations to what the blog post I linked to says.

joearkay, 2020-02-17:
@todbot - that's super interesting. Thanks for the nudge.

todbot, 2020-02-17:
It's really frustrating. I've taken to exclusively using `nvm` to manage node/npm versions.