node-hid/node-hid #366: Uncaught Error: cannot open device

Swiftb0y, 2020-03-29:
I am having trouble opening a HID device on Linux.
Usb enumeration:
```javascript
{
    vendorId: 11123,
    productId: 5,
    path: '/dev/hidraw1',
    serialNumber: '',
    manufacturer: 'Pioneer DJ Corporation',
    product: 'CDJ-2000NXS2',
    release: 389,
    interface: 4
},
```
Opening via path and vid/pid both fails with the exact same error:
```
Uncaught Error: cannot open device with vendor id 0x2b73 and product id 0x5
at new HID (/home/swiftb0y/Sourcerepositories/cdjhidjs/node_modules/node-hid/nodehid.js:49:17)
```
Starting node as root works as a temporary workaround.
I'm reasonably sure that it is not a permission issue:
I've created a udev entry (`SUBSYSTEM=="usb", ENV{DEVTYPE}=="usb_device", ATTRS{idVendor}=="2b73", TAG+="uaccess"`) and also got it to work without root permissions using in python using pyhidapi.

todbot, 2020-03-29:
Hi,
If it works as "root" but not as your normal user, then it's almost definitely a udev issue.  Once you have the udev rules set up, unplug & replug your device, and try again.

`node-hid` and `pyhidapi` use the same underlying C library (`hidapi`) so if one works the other should too.

Swiftb0y, 2020-03-29:
I did a bunch of more test (I also restarted the PC in the meantime) but it still doesn't work. I'm not sure how to troubleshoot this further as well.
Could it be device related or an issue with hidapi? When I successfully connect to the device using python, read some data, disconnect and try to reconnect, it still succeeds in reconnecting, but I waits indefinitely when reading data. The device also disappeared from `HID.devices()` while its still present when using pyhidapi's `hid.enumerate()`.

pablogravielseo, 2020-09-21:
Hi, this is happening for me also, I'm using a Raspberry Pi 4 with Node.js version 12.18.4 trying to read the data from an RFID reader and I got the following response: 

```
{
  vendorId: 65535,
  productId: 53,
  path: '/dev/hidraw0',
  serialNumber: '08FF20150112',
  manufacturer: 'Sycreader',
  product: 'USB Reader',
  release: 256,
  interface: 0
}
```

`Error: cannot open device with path /dev/hidraw0`

todbot, 2020-09-21:
@pablogravielseo, have you installed the udev rules for your device, or are running your process as the `root` user?  Or, if this device, acts like a keyboard, have you uninstalled the keyboard driver for this vid/pid?

pablogravielseo, 2020-10-03:
Was a "udev rules" issue, I've added myself to the "plugdev" group following this article https://unix.stackexchange.com/questions/85379/dev-hidraw-read-permissions/85459, now I can read the data, thank you @todbot!

closed 2020-10-04 by todbot

AckerApple, 2021-04-27:
@jongmoseo your link was most extra useful to me this evening, thank you very so much

jongmoseo, 2021-04-27:
Hi @AckerApple, great, no problem, I'm glad that it was useful for you :)