node-hid/node-hid #367: [Electron] Cannot open device with

saifeiLee, 2020-04-13:
## Description

I can get and open the device by js script running in terminal. Also, In Electron **render-process**, I can get the devices info as expected.

[image]

But when I call `const device = new HID.HID(item.vendorId, item.productId);` , I failed. Console prints the error: 
```javascript
Uncaught Error: cannot open device with vendor id 0x1ff7 and product id 0xf05
```
**Screenshot**:
[image]

todbot, 2020-04-13:
Hi,  a few questions:
- What does the entirety of `HID.devices()` look like?  It may be the device is a multi-usage device, meaning it shows up as multiple devices, made distinct by usages or interface numbers. 
- Are you opening the device somewhere else at the same time?
- Can you copy that exact code snipping and run it in the Electron main process?  
- Can I see the output of that exact code snipping running in Node?

Also in general you should be filtering by vendorId and productId, not manufacturer strings as they can change or not be portable across OSes.

saifeiLee, 2020-04-14:
> Hi, a few questions:
>
> * What does the entirety of `HID.devices()` look like?  It may be the device is a multi-usage device, meaning it shows up as multiple devices, made distinct by usages or [...]

Thanks for your reply. 
As you suggested, I found the device has been opening elsewhere. after closing, issue  fixed. 
Besides, Thanks for your suggestion about filtering device by vendorId / productId . It's really helpful. 😁❤️

closed 2020-04-14 by saifeiLee