node-hid/node-hid #368: Uncaught Error, the module was compiled against a different Node.js version using...

lukebayes, 2020-04-15:
```bash
node_modules/node-hid/build/Release/HID-hidraw.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 76. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
```

Here are some environmental details:
```
Node v12.16.2
Electron v8.2.2
npm v6.14.4
Ubuntu 19.10
```
I've tried running `npm rebuild`, `npm rebuild node-hid` and './node_modules/.bin/electron-rebuild`.

Alas, still not working.

lukebayes, 2020-04-15:
Fixed in a few steps:

1) Added the following to my Electron application entry point:
```javascript
console.log("Launching Electron with:");
console.log(process.versions);
```
This showed the following from Electron
```bash
Launching Electron with:
{
  node: '12.13.0',
  v8: '8.0.426.30-electron.0',
  uv: '1.32.0',
  zlib: '1.2.11',
  brotli: '1.0.7',
  ares: '1.15.0',
  modules: '76',
  nghttp2: '1.39.2',
  napi: '5',
  llhttp: '1.1.4',
  http_parser: '2.8.0',
  openssl: '1.1.0',
  icu: '65.1',
  unicode: '12.1',
  electron: '8.2.2',
  chrome: '80.0.3987.163'
}
```
2) I downgraded my nodejs version to match the one that's packaged with Electron (v12.13.0)
3) Reinstalled node modules, including Electron, still didn't quite work.
4) Ran `electron-rebuild node-hid`

It worked!

closed 2020-04-15 by lukebayes

todbot, 2020-04-15:
Hi @lukebayes,
Glad you figured it out!  As you realized, it was the last step `electron-rebuild` that does the actual magic.  The Node that's inside Electron is distinct (even the same "version" is different under the hood, that's what `NODE_MODULE_VERSION` is hinting at), so `electron-rebuild` essentially does a `npm install` for any native modules it finds that need to be rebuilt.  It doesn't matter what version Node you're running.