node-hid/node-hid #369: 'cannot open device with vendor id' when new device plug in

saifeiLee, 2020-04-16:
## Description

I plug in a new device and using `usb-detect` to monitor the changes. As expected, I can detect new device pluged in.
[image]
And I want to open the new HID device with its vid/pid. But I got the error.
```javascript
Uncaught Exception:
Error: cannot open device with vendor id 0x1ff7 and product id 0xf05
    at new HID (/Users/waltermitty/Code/cvte/electron-easi-camera/app/node_modules/node-hid/nodehid.js:49:17)
```
I filtered the device list which returned by `HID.devices()` with the pid/vid. It's empty.
[image]
Is there anything I went wrong?
If you need and further info, please let me know and I will append as soon as possible.

todbot, 2020-04-16:
The device isn't being opened by another process, like before. :)

But really I think the problem may be that the `devices` variable is stale.  You need to do `devices = HID.devices()` to pick up the updated device list after the device insertion.  With the code you quote, `devices` contains the list of HID devices at the start of the code, not when `add` was called.

saifeiLee, 2020-04-16:
> The device isn't being opened by another process, like before. :)
>
> But really I think the problem may be that the `devices` variable is stale. You need to do `devices = HID.devices()` to pick up the [...]

Pretty sure the device not been opened. lol
 In fact, I tried to call `HID.devices()` when device added. The weirdest part is the new-added device is not in the updated list.

saifeiLee, 2020-04-16:
> The device isn't being opened by another process, like before. :)
>
> But really I think the problem may be that the `devices` variable is stale. You need to do `devices = HID.devices()` to pick up the [...]

Hah.. Pretty sure the device not been opened.
In fact, I tried to call HID.devices() when device added. The weirdest part is the new-added device is not in the updated list.

todbot, 2020-04-16:
Depending on the OS, you will not see the device immediately on usb insertion.  It takes time for the OS to hook up the HID driver to the USB device (this can take more than a second sometimes I've seen).  

Try doing a `setTimeout(function(){console.log(HID.devices()},3000)` in your `.on('add')` and see if you see the device then.

saifeiLee, 2020-04-17:
well. You're right. It is OS delay. The reason for the first time I tried
`setTimeout` but failed, is that I did not use `HID.devices()`. emmm.. What
can I say? hah.
Anyway, thank you so much! May you have a good day!

closed 2020-04-17 by saifeiLee

saifeiLee, 2020-04-17:
> Depending on the OS, you will not see the device immediately on usb insertion. It takes time for the OS to hook up the HID driver to the USB device (this can take more than a second sometimes I've [...]

Well. You're right. It is OS delay. This time I tried `setTimeout` works. But last time failed. 
Anyway, thank you so much! May you have a good day!