node-hid/node-hid #370: Cannot open device on Debian

saifeiLee, 2020-04-20:
## Description
I packaged my electron app to `.deb` and installed on debian. 

#### Expected
Open device.

#### Fact
```javascript
{
  constructor: 'Error',
  stack: 'Error: cannot open device with vendor id 0x1ff7 and product id 0xf05\n' +
    '    at new HID (/home/cvte/Code/electron-easi-camera/app/node_modules/node-hid/nodehid.js:49:17)\n' +
    '    at forEach (/home/cvte/Code/electron-easi-camera/app/utils/hid-function.ts:183:22)\n' +
    '    at Array.forEach (<anonymous>)\n' +
    '    at openAllMCU (/home/cvte/Code/electron-easi-camera/app/utils/hid-function.ts:182:14)\n' +
    '    at Timeout._onTimeout (/home/cvte/Code/electron-easi-camera/app/module/usb-detection.ts:25:11)\n' +
    '    at listOnTimeout (internal/timers.js:531:17)\n' +
    '    at processTimers (internal/timers.js:475:7)'
}

```
The app works perfectly on os x.

I don't know how to debug this error. Could you give me some advice?I know the context is limited. If you need any further info, please let me know and i will append as soon as possiable.

todbot, 2020-04-20:
Hi,
Please review the permission requirements for Linux:
https://github.com/node-hid/node-hid#udev-device-permissions
On Linux you need to give the user permission to use the device

saifeiLee, 2020-04-21:
> Hi,
> Please review the permission requirements for Linux:
> https://github.com/node-hid/node-hid#udev-device-permissions
> On Linux you need to give the user permission to use the device

Thanks for your suggestion. I solved it by adding rules.d file.Thank you again. Wish you have a good day!

closed 2020-04-21 by saifeiLee