node-hid/node-hid #372: Trying to read from HID IN endpoint

koalazak, 2020-04-28:
Hello,
I know nothing about USB/HID...But Im trying to get data from a HID device. I sniffed the USB trafic with wireshark and I saw the data what I want. Also I saw the packet of interest is a "reply" to a packet sent by the host.
So Im trying to emulate the initiator host's packet to get the data packet of my interest.

This is the packet that the host in sending:
```
Frame 5454: 27 bytes on wire (216 bits), 27 bytes captured (216 bits) on interface 
USB URB
    [Source: host]
    [Destination: 1.7.4]
    USBPcap pseudoheader length: 27
    IRP ID: 0xffffd285cba7b8a0
    IRP USBD_STATUS: USBD_STATUS_SUCCESS (0x00000000)
    URB Function: URB_FUNCTION_BULK_OR_INTERRUPT_TRANSFER (0x0009)
    IRP information: 0x00, Direction: FDO -> PDO
    URB bus id: 1
    Device address: 7
    Endpoint: 0x84, Direction: IN
        1... .... = Direction: IN (1)
        .... 0100 = Endpoint number: 4
    URB transfer type: URB_INTERRUPT (0x01)
    Packet Data Length: 0
    [Response in: 5461]
    [bInterfaceClass: HID (0x03)]
```
in hex:
```
0000   1b 00 a0 b8 a7 cb 85 d2 ff ff 00 00 00 00 09 00   ................
0010   00 01 00 07 00 84 01 00 00 00 00                  ...........
```

So, the host sent that packet and a few frames later I see a packet from device to host with the information.
How Can I emulate this packet with node-hid? Seems like I should use `write()`. Or I should use `getFeatureReport()`? or just .read() ?

this is my current code:
```js
var HID = require('node-hid');
var device = new HID.HID(path);

device.on("data", function(data) {
  console.log(data);
});

device.on("error", function(err) {
  console.log('error', err);
});

device.read(function (e, d){
	console.log(e,d);
})

setInterval( function() {
    device.write([]); // I have no idea what I should send here! the endpoint address?
}, 1000);
```

And this is the interface descriptior:
```
Interface #2 - HID   
            Alternate Setting   0
            Number of Endpoints   1
            Interface Class:   3   (HID)
            Interface Subclass;   0
            Interface Protocol:   0
            HID Descriptor   
                Descriptor Version Number:   0x0111
                Country Code:   0
                Descriptor Count:   1
                Descriptor 1   
                    Type:   0x22  (Report Descriptor)
                    Length (and contents):   28
                        Raw Descriptor (hex)    0000: 06 AB FF 0A 00 02 A1 01  75 08 15 00 26 FF 00 95  
                        Raw Descriptor (hex)    0010: 40 09 01 81 02 95 40 09  02 91 02 C0 
                    Parsed Report Descriptor:   
                          Usage Page    (Vendor defined 171) 
                          Usage 512 (0x200)    
                              Collection (Application)    
                                Report Size.............    (8)  
                                Logical Minimum.........    (0)  
                                Logical Maximum.........    (255)  
                                Report Count............    (64)  
                                Usage 1 (0x1)    
                                Input...................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Bitfield) 
                                Report Count............    (64)  
                                Usage 2 (0x2)    
                                Output..................   (Data, Variable, Absolute, No Wrap, Linear, Preferred State, No Null Position, Nonvolatile, Bitfield) 
                              End Collection     
            Endpoint 0x84 - Interrupt Input   
                Address:   0x84  (IN)
                Attributes:   0x03  (Interrupt)
                Max Packet Size:   64
                Polling Interval:   1 ms
```

thank you in advance!

todbot, 2020-04-28:
Hi, This is good research!  For HID the most important bit of data is the HID Descriptor.  On Mac & Windows this is hard to get, but if you're on Linux, you can run `usbhid-dump` (it's part of usbutils: https://github.com/gregkh/usbutils )  I believe it's installed by default on most Debian-based systems. If not `sudo apt install usbutils`.

Here's more info on `usbhid-dump`: https://github.com/gregkh/usbutils/tree/master/usbhid-dump  The command `usbhid-dump` gives you just the bytes of the HID Descriptor, you can then feed it into `hidrd-convert` or use an online tool like https://eleccelerator.com/usbdescreqparser/  to parse the bytes into the standard human-readable form.

The HID descriptor tells what HID commands the device is expecting and what it emits. From there you can know what byte arrays to emit and receive in `node-hid`.

koalazak, 2020-04-28:
thank you! now I have the HID report descriptor:
```
0x06, 0xAB, 0xFF,  // Usage Page (Vendor Defined 0xFFAB)
0x0A, 0x00, 0x02,  // Usage (0x0200)
0xA1, 0x01,        // Collection (Application)
0x75, 0x08,        //   Report Size (8)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x95, 0x40,        //   Report Count (64)
0x09, 0x01,        //   Usage (0x01)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x40,        //   Report Count (64)
0x09, 0x02,        //   Usage (0x02)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0xC0,              // End Collection

// 28 bytes```

but I do not undestund what I should send with write()...

On my sniff data I also see a frame like "SET REPORT REQUEST"
```
Setup Data
    bmRequestType: 0x21
    bRequest: SET_REPORT (0x09)
    wValue: 0x0200
    wIndex: 2
    wLength: 64
    bRequest: 9
    wValue: 0x0200
    wIndex: 2 (0x0002)
    wLength: 64
```

what it means? Should I sent that? those bytes in that order with `.write()` ?
sorry im very lost...

koalazak, 2020-04-28:
so, is my undestanding right? I must "write()" a command to get data back?
And what im sending with write()? a descriptor too? what write() does?
thank you

todbot, 2020-04-28:
The logic of what data packets (known as "reports" in USB HID parlance) is sent by the device depends on the device's firmware, it's not something dictated by USB or HID. For instance, a USB mouse continuously sends HID reports without the host sending any HID reports to it. And a USB keyboard also will send HID reports unprompted when a user presses a key, but the host can also send a report to the keyboard to change the capslock/numlock/scrolllock LEDs.  (These devices are both "owned" by the OS so unfortunately you can't normally see these reports using `node-hid`)

From your HID report descriptor, the "Usage Page" is 0xFFAB, meaning it's a vendor-defined device and thus not taken by the system (for comparison, Usage Page 0x01 is keyboard/mouse).  And your device communicates via 8-bit bytes (REPORT_SIZE)  and 64-byte packets (REPORT_COUNT) and that there are no report Ids.  This means you will be sending & receiving 64-byte buffers to the device.

That makes me think what you're seeing the host send isn't the whole story.  What kind of device is this?

koalazak, 2020-04-29:
Ok, I think I need to read again all this paper https://www.usb.org/sites/default/files/documents/hid1_11.pdf and then come back with questions :p

The device is a miniPCR machine. I see with wireshark all the 64 bytes that the device is sending to host. And I was able to decode the format of the data (it is a google's protobuf structure).
But I don't kown how to "ask" to the device to get those 64 bytes via external node.js program.
I thought that having all the traffic it would be easy to get that data from another external node.js program, but seems like it doesn't.

When I plug de device I see a few packets. But when I open the User Inferface program to control de device I saw some initialization packets and then the the packets of interest start arraiving every second, with all the data (temperature, cicles, etc).
https://ibb.co/S7rvKFZ

I will research more...
and close this issue (because it is not an issue with the lib)

thank you again for your time!

closed 2020-04-29 by koalazak

todbot, 2020-04-29:
You're pretty close.  From your wireshark screenshot, you want to pay attention to the SET_REPORT (send data from host to device) and GET_REPORT (send data from device to host) packets.  

Once you find out what initialization reports to send to the device, and if the device just keeps sending data after that, the node-hid code becomes something like:
```
var device = HID.HID( vid,pid );
device.on('data', function() {  console.log(data) }); // print on report received
var init_report = ['w','h','a','t','e','v','e','r',0xCA,0xFE]; // whatever the bytes to send are
device.write(init_report);
setTimeout( function() { console.log("waiting for data"); }, 5000);
```

koalazak, 2020-04-29:
thank you! So I need to take care of sending SET_REPORT packets and node-hid take care of get the interrupt and read the info from the endpoing when it is there, right?

I will try with the SET_REPORTs that I see, but I have a doubt...
https://ibb.co/0GgSWvS
this is one packet ☝️ 

`write()` function is expecting only the "data fragment" bytes (in blue), right? 
I don't need to worry about the "Setup Data" bytes (in red), right? . That is something node-hid handles for me (wIndex, wValue, bRequest, etc)?

Regarding the `data fragment` length in blue (64 bytes) I think this is what I need to send. But I ask just to confirm...

Also, what are the `0xCA,0xFE` bytes that you add to the tail? I don't see those bytes in my campture data in blue part.

thank you again

koalazak, 2020-05-07:
Hello @todbot sorry to bother you again. But do you ahve any spike about my last comment? Im still figthing with this.

thank you

todbot, 2020-05-07:
Apologies, I don't have anything more to add until you figure out the HID reports required by the device.
But yes, you do not need to worry about any of the normal USB setup, that is all handled by the OS when `node-hid` (or any HID-using program) opens the device and tries to send or receive HID reports