node-hid/node-hid #374: Electron 9 error message

Sergey-Churilin, 2020-05-12:
### Description
I'm getting an error message in console **Uncaught Error: Loading non-context-aware native module in renderer**

If use a workaround provided by electron to be possible to use currently such modules, it replaces with warning: **(node:15876) Electron: Loading non-context-aware native module in renderer**
But in Electron 11 this workaround will be removed by Electron and usage of spellchecker module will be impossible

### Steps to Reproduce
https://github.com/Sergey-Churilin/Electron-samples/tree/node-hid
1. Run npm i
2. Run npm start

**Expected behavior:**

No error or warnings in console, application should work

**Actual behavior:**

If not to use a workaround, application doesn't work correctly

**Reproduces how often:**

100%

### Versions
Node version: 12.16.1

### Additional Information
Discussions about how the issue can be solved: https://github.com/electron/electron/issues/18397

todbot, 2020-05-12:
Thanks for the detailed write-up. I'm looking into this now and hopefully the change to `node-hid` is minimal

Sergey-Churilin, 2020-07-07:
Hi @todbot , was the issue fixed in v1.3.0?

todbot, 2020-07-07:
Hi!  node-hid has been bandaged for this issue.  I did the minimal to make the deprecation errors go away and let you run Electron without doing `app.allowRendererProcessReuse=false`.  

However, by the underlying `hidapi` C-library is not thread-safe.   Solving this much larger problem is more difficult, and goes beyond my current understanding of `nan` and `v8`.

For now, please make sure all your `node-hid` calls are in a single thread.

I've documented some of this here:
https://github.com/node-hid/node-hid#thread-safety-worker-threads-context-aware-modules

doug-a-brunner, 2022-04-05:
Now that `node-hid` is using N-API, can this issue be closed?

todbot, 2022-04-05:
The underlying `hidapi` library that `node-hid` wraps is not necessarily thread-safe.

doug-a-brunner, 2022-04-05:
OK, so the change to N-API didn't lead to creating multiple library contexts as would be needed to close this out. Thanks for the clarification.