node-hid/node-hid #376: Electron 9.0.2: No prebuilt binaries found

Sergey-Churilin, 2020-06-10:
### Description
I faced with issue that during building an electron application I see such warnings:

**For windows**

    prebuild-install http request GET https://github.com/node-hid/node-hid/releases/download/v1.2.0/node-hid-v1.2.0-electron-v80-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/node-hid/node-hid/releases/download/v1.2.0/node-hid-v1.2.0-electron-v80-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=9.0.2 runtime=electron arch=x64 libc= platform=win32)

**For Linux**

    prebuild-install http request GET https://github.com/node-hid/node-hid/releases/download/v1.2.0/node-hid-v1.2.0-electron-v80-linux-x64.tar.gz
    prebuild-install http 404 https://github.com/node-hid/node-hid/releases/download/v1.2.0/node-hid-v1.2.0-electron-v80-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=9.0.2 runtime=electron arch=x64 libc= platform=linux)

**Expected behavior:**
Binaries are located under the path

**Actual behavior:**
Binaries not found

**Reproduces how often:**
100%

**Versions:**
OS: Windows 10, Ubuntu 16.04
Node version: 10.16.0
Electron version: 9.0.2
Node-hid version: 1.2.0

todbot, 2020-08-13:
This is for an old version of node-hid. Current version is 1.3.0

todbot, 2020-08-13:
At the time this issue was opened, `node-hid@1.2.0` was latest release and released before Electron 9 (thus no prebuilds). As of `node-hid@1.3.0`, Electron 9 prebuilds exist.

closed 2020-08-13 by todbot

astellingwerf, 2021-03-07:
The same issue still /re-exists for `node-hid@2.1.1`:
```
  ⨯ cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible
                                    cause=exit status 1
                                    errorOut=prebuild-install info begin Prebuild-install version 6.0.0
    prebuild-install info looking for cached prebuild @ /home/astellin/.npm/_prebuilds/1d45c9-node-hid-v2.1.1-electron-v80-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/node-hid/node-hid/releases/download/v2.1.1/node-hid-v2.1.1-electron-v80-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/node-hid/node-hid/releases/download/v2.1.1/node-hid-v2.1.1-electron-v80-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=9.4.0 runtime=electron arch=x64 libc= platform=win32)

                                    command=/usr/bin/node /mnt/c/git/companion/node_modules/prebuild-install/bin.js --platform=win32 --arch=x64 --target=9.4.0 --runtime=electron --verbose --force
                                    workingDir=/mnt/c/git/companion/node_modules/node-hid
```

todbot, 2021-03-07:
@astellingwerf I assume you are using `electron-rebuild` or equivalent?  It's asking for the incorrect binary. It's asking for `node-hid-v2.1.1-electron-v80-win32-x64.tar.gz` and it should be asking for `node-hid-v2.1.1-napi-v3-win32-x64.tar.gz`

I believe this is a problem with `electron-rebuild`. I've opened in issue here:  https://github.com/electron/electron-rebuild/issues/520
Please ask them about it.