node-hid/node-hid #379: Cannot get devices

toanhda, 2020-06-28:
I was try:
var HID = require('node-hid');
var devices = HID.devices();
But error:
App threw an error during load
Error: Module did not self-register: '/Users/toan/personal/electron-quick-start/node_modules/node-hid/build/Release/HID.node'.
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034:18)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:815:32)

OS: macos
    "electron": "^9.0.5"
    "node-hid": "^1.3.0"

Please help me

todbot, 2020-06-28:
Can you get this project to work?  https://github.com/todbot/electron-hid-test

toanhda, 2020-07-09:
@todbot  thank you for your support

closed 2020-07-09 by toanhda