node-hid/node-hid #380: HID Report Descriptor, how

AdamChlupacek, 2020-07-07:
Hello, 

This is more of a question about usage of the library.

We have an application that is using HID head phones for controlling of a call, currently we have static masks for events we are sending to the headphones, as such we can only support the devices which for which we have hardcoded mask in our application. 

I wanted to extend this function for generic headphones and according to the HID spec, the headphones should have Report Descriptor in them, I managed to read the descriptor on macOs (I assume linux will be the same) using libusb via the control transfer. But I seem to be failing to find any way to read the Report Descriptor on Windows.

Is there any suggested way how we should find out what reports to send / receive from the HID devices with this library?

todbot, 2021-03-01:
Hi,

To help with this kind of research, I've created two tools:

- https://github.com/todbot/win-hid-dump
- https://github.com/todbot/mac-hid-dump

You can use the output of either (and the original Linux tool https://github.com/DIGImend/usbhid-dump) to feed into a HID Report Descriptor parser like https://eleccelerator.com/usbdescreqparser/

Leaving this issue open for now to remind me to add this to the README. We can't add this to `node-hid` but these tools have proved useful when researching HID devices.

yuchunli, 2021-08-21:
@AdamChlupacek Curious whether you figured out how to extend your function for generic headphones?

mcuee, 2023-05-06:
mac-hid-dump works fine but not win-hid-dump due to the limitation of hidsharp library it uses.

For now, the best tool is actul the latest hidtest from hidapi git master.

Ref:
* https://github.com/libusb/hidapi/pull/451
* https://github.com/libusb/hidapi/issues/249