node-hid/node-hid #381: How to: read data on in pin

Xeobass, 2020-07-16:
Hi,

i have a question about how to read data received form a specified in pin.

i have this usb relay board ( https://denkovi.com/usb-relay-board-four-channels-for-home-automation-v2 ) i can send data correctly to open/close relay (1-4) and open/close in pins (4-8) but i can't read the state or any kind of data.  If i unplugge the usb the device.on(error.(e)=>{...}) shows an error but...the device.on(data,(d)=>{...}) is always empty.....

someone used this kind of usb relay board before? how i could read in pin data and/or relay state?

Thanks!

todbot, 2020-07-16:
This is not really a `node-hid` issue, but really a question about this relay board. 
Does the demo code/apps listed on the page you reference work the way the way you expect and can do what you're trying to do in `node-hid`?
Without knowing the HID Report Descriptor for this device, it's almost impossible to provide guidance on how to use it `node-hid`.

todbot, 2020-07-16:
But from looking at this tech note for the MCP2200 chip on that board: http://ww1.microchip.com/downloads/en/devicedoc/93066a.pdf
It looks like you may be able to do a `READ_ALL` command to get the data back from the device with something like:
```
var device = new HID.HID(....) // fill in 

// print out any IN reports received
device.on('data', function(data) {
    console.log("got data:", data.toString('hex') );
});

// send "READ_ALL" command
var REPORT_ID = 0   // assume there's no report ids
var REPORT_LENGTH = 16  // from MCP2200 datasheet
var report = new Array(REPORT_LENGTH + 1).fill(0)   // + 1 for reportId
report[0] = REPORT_ID // reportId or zero if none
report[1] = 0x80  // 'READ_ALL' command
var rc = device.write(report);

console.log("Waiting 10 seconds for data");
setTimeout( function() {
    console.log("Done");
    device.close();
}, 10000);

```

The above is assuming it's using IN and OUT reports.  If it's using FEATURE reports, then the code would look something like:
```
// send "READ_ALL" command
var REPORT_ID = 0 
var REPORT_LENGTH = 16
var report = new Array(REPORT_LENGTH + 1).fill(0)   // + 1 for reportId
report[0] = REPORT_ID // reportId or zero if none
report[1] = 0x80  // 'READ_ALL' command
var rc = device.sendFeatureReport(report);
var result = device.getFeatureReport(REPORT_ID, REPORT_LENGTH+1)
console.log("got data:", result.toString('hex') );

```

Xeobass, 2020-07-17:
thanks todbot! it worked!

it is something similar what i was doing but my "command array" was wrong, i wrote 0x01 0x80 instead 0x80 only (among other isues with a listener)

if someone have a similar question: 

 the data recieved is:

                                                                *
<Buffer 80 00 0a 00 00 00 00 00 02 70 08 0d 60 97 32 c5>

the 10th position indicates the state of the relays (08 hex = 08 dec = 00001000 bin => the 4th relay is active)

<Buffer 80 00 0a 00 00 00 00 00 02 70 ff 0d 60 97 32 c5>
ff hex = 255 dec = 11111111 bin => all relays are active

Thank you so much again todbot!

closed 2020-07-17 by Xeobass