node-hid/node-hid #382: Module did not self-register

20PercentRendered, 2020-07-16:
Hi, i am creating an app with nodegui, and i've added node-hid and elgato-stream-deck. elgato-stream-deck was giving me issues, so i decided to add the following to the start of my index.js
`const HID = require("node-hid");

HID.setDriverType('libusb');
HID.devices()
`
I am using webpack, so i added the following to the webpack.config.js
` externals: {
    "node-hid": 'commonjs node-hid'
 }`
Everything works fine, until this error occurs:
`Error: Module did not self-register: '/home/a/Documents/Projects/nodedecks/node_modules/node-hid/build/Release/HID.node'.
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1226:18)
    at Module.load (internal/modules/cjs/loader.js:1040:32)
    at Function.Module._load (internal/modules/cjs/loader.js:929:14)
    at Module.require (internal/modules/cjs/loader.js:1080:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at bindings (/home/a/Documents/Projects/nodedecks/node_modules/node-hid/node_modules/bindings/bindings.js:112:48)
    at loadBinding (/home/a/Documents/Projects/nodedecks/node_modules/node-hid/nodehid.js:21:46)
    at Object.showdevices [as devices] (/home/a/Documents/Projects/nodedecks/node_modules/node-hid/nodehid.js:127:5)
    at eval (webpack:///./src/index.js?:8:5)
    at Object../src/index.js (/home/a/Documents/Projects/nodedecks/dist/index.js:3319:1)`

todbot, 2020-07-16:
It sounds like you are using `node-hid` in an Electron app? If so, you need to use `electron-rebuild` or `electron-builder` to rebuild `node-hid` for Electron.  See: https://github.com/node-hid/node-hid#electron-projects-using-node-hid

todbot, 2020-07-16:
Oh sorry, I didn't see you were using NodeGUI.  It looks like for NodeGUI, you need to follow the instructions here: https://docs.nodegui.org/docs/guides/using-native-node-modules/

20PercentRendered, 2020-07-16:
That didn't seem to change anything. It is still giving me the same error.

todbot, 2020-07-16:
Which "that"? You need to be more specific.  Do you have a Node-only version of your code that works so you can eliminate any issues with NodeGUI?  What OS are you on?  The "libusb" backend is only for Linux.

20PercentRendered, 2020-07-16:
I meant following the instructions on rebuilding didn't seem to do anything. It works on the electron demo app, mentioned in #379 . 
Running Arch Linux.

20PercentRendered, 2020-07-17:
Fixed by building qode from source and then building node-hid from source with the specified node build.
So, for future readers,
1. Create a temporary directory for your custom qode build, and cd into it.
2. Clone qode into it.
3. Install submodules.
4. `node build.js`
Once built, go back to your project root, run
`npm rebuild node-hid --nodedir=/yourproject/thetempdir/qode/node --build-from-source`
It should now work.

closed 2020-07-17 by 20PercentRendered

todbot, 2020-07-17:
Dang, that is a pretty "nuke from orbit" kind of solution.  How did you discover that was the correct approach?  Thank you for your research. The NodeGUI team might want to know about this too so they can update their documentation.