node-hid/node-hid #383: Error:could not read from HID device

ashtam55, 2020-07-23:
Stuck in for 2 days Please Help if possible,
Having trouble in getting data from the HID device which is a webkey button. I am able to get device info data given below. 
```
{
    vendorId: 1452,
    productId: 523,
    path: '\\?\hid#vid_05ac&pid_020b#8&2fbc1187&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd',
    manufacturer: 'WebKey',
    product: 'WebKey',
    usagePage: 1,
    usage: 6
  }
```

When I connect to it and start listening for data it throughs an error 
```
[31092:0723/160621.151:INFO:CONSOLE(28)] "vid 16700", source: C:\Users\Saumya\electron-hid-toy\scripts\app.js (28)
[31092:0723/160621.153:INFO:CONSOLE(91)] "Only showing interested device", source: C:\Users\Saumya\electron-hid-toy\scripts\app.js (91)
[31092:0723/160621.403:INFO:CONSOLE(76)] "device error :  Error: could not read from HID device", source: C:\Users\Saumya\electron-hid-toy\scripts\app.js (76)
```

My code snippet
```
var HID = require('node-hid');

var devices = HID.devices();
console.log("node-hid loaded: devices:", HID.devices());

var deviceInfo = devices.find( function(d) {
    var isWebkey = d.vendorId===0x05AC;
    return isWebkey;
});
if( deviceInfo ) {
  var device = new HID.HID(deviceInfo.path);

  device.on('data',function(data){
    console.log("device data : ",data);
    document.body.style.backgroundColor = "green";
    setTimeout(() => { document.body.style.backgroundColor = "white"; }, 1000);
  });

  device.on('error',function(err){
    console.log("device error : ",err);
  });

}
```
My node version ```v12.18.2``` 
Electron version ```8.4.0``` also tried on ```9.0.5``` but not working.
node-hid version ```1.3.0```

I just need to get an event whenever button is pressed. Please suggest me a workaround.
Thank you

todbot, 2020-07-23:
Without running any application, what happens if you press the Webkey button?  Does it "type" into your computer?  If so, then it is a keyboard to the OS and node-hid cannot access it.

todbot, 2020-07-23:
Oh yes, it's usagePage 1 (Generic Desktop) and usage 6 (Keyboard), so `node-hid` definitely cannot access it unless you unload the OS HID keyboard driver for this device, which I don't think is possible in Windows.  You can find details about HID Usages here: https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf

Sorry about that.  Instead, you can use standard key event techniques to read the keyboard if your app is focussed and use that to determine if the Webkey is pressed.

closed 2020-07-23 by todbot

ashtam55, 2020-07-26:
@todbot Thanks for your suggestions and helping me out. I will try to use it as a standard keypress event using the keyboard button press event library and update you here.
Thanks Again 👍