node-hid/node-hid #384: Add support for NodeJS 14.

MicahZoltu, 2020-07-25:
NodeJS 13 is official end of life (out of maintenance) and NodeJS 14 is now Current.  Recommend adding NodeJS 14 to the prebuilt binaries list.

Specifically for my use case this setup is failing:
```
No prebuilt binaries found (target=14.4.0 runtime=node arch=x64 libc= platform=win32)
```

todbot, 2020-07-25:
Are you using the latest version of `node-hid`?  I'm seeing prebuild `node-hid-v1.3.0-node-v83-win32-x64.tar.gz` in the release.

`node-hid` is at the mercy of `prebuild`, which in tun is at the mercy of `node-abi`.  We build Mac & Windows prebuilds for every version reported by `node-abi`.  In current `node-abi`, it says Node v14 is supported and identified as abi 83.  

Perhaps either Node v14.4.0 is a different ABI or you're not using `node-hid@1.3.0`?

MicahZoltu, 2020-07-25:
Digging further, it appears I was getting node-hid through as a transitive dependency that wasn't up to date.  Sorry for the false alarm!

closed 2020-07-25 by MicahZoltu

todbot, 2020-07-26:
No problem, it happens all the time.
Which project were you using that uses `node-hid`?  Just curious to see what people are using it for.

MicahZoltu, 2020-07-26:
Integration with a hardware wallet for crypto-currencies: https://github.com/LedgerHQ/ledgerjs