node-hid/node-hid #385: Node-hid can not connect 2 RFID125 Khz usb reader at same time due to same VID and PID

tranduchau081188, 2020-07-30:
Dear all,
I build the Pike Parking project that require 2 RFID125 Khz usb reader on Node-red and Raspi 3B.
I install Node-red-contrib-usbhid however usbhid detect same VID and PID for both RFID125 Khz usb reader, only serial is defferent.
Are there any way for usbhid to see both RFID125 Khz usb reader in 1 Raspi 3B?
Thank You!

todbot, 2020-07-30:
Most RFID readers show up as a keyboard.  These devices cannot be read by `node-hid`, as the OS owns keyboards.  Use standard keyboard-reading techniques to read these readers.  If the reader has a "USB serial" mode, then use `node-serialport`.  If the reader has a "HID data" mode, then you may be able to use `node-hid`, if you know the protocol of the HID data mode.

todbot, 2020-07-30:
Howver, if you are asking how to distinguish different HID devices when the VID and PID are the same, but some other aspect is different, then you should use `HID.devices()` to get a list of devices and then pick from the list that is returned.

You can see an example of filtering by information other than VID/PID in a few of the examples, like: https://github.com/node-hid/node-hid/blob/master/src/test-teensyrawhid.js

tranduchau081188, 2020-07-30:
hi @todbot ,
Thank for your answer.
however, currently my system working on Node-red.
I cannot use Node.hid() alone. May we can config to add other aspect of HID (ex: unique serial is best choice)

todbot, 2020-07-30:
I am unclear what you are asking. It sounds like you need to ask the node-red team.

closed 2020-11-24 by todbot