node-hid/node-hid #386: Error: could not read from HID device

patrykscheffler, 2020-08-05:
I was able to connect and read data from WD-678XW 2D Barcode Scanner using `node-hid` in macOS. However in Windows 10 when I connect to the device and start listening for data it throws an error 'Error: could not read from HID device'.

Is it possible to read data from this device on Windows 10? In the application, I also used standard key event techniques to read the keyboard, but I would like to use a barcode scanner even when the app is unfocused.

patrykscheffler, 2020-08-24:
I also tried to connect FT260 which is an HID class device and doesn't appear as an HID keyboard in windows.
And again there is no issue on macOS with communication but on Windows 10 there is the same issue as above. 
I can open the device but cannot read or write to it. 

https://www.ftdichip.com/Products/ICs/FT260.html

I am using node-hid version 1.3.0

todbot, 2020-08-24:
Perhaps there a driver on Windows that is interfering with standard HID access?  You should be able to see if there's any custom drivers installed for a device by find the device in the Windows Device Manager, right-clicking on it and selecting "Properties" and then the "Driver" tab.

rasgo-cc, 2021-01-28:
I have the same result. Not sure where to look at. I tried two different barcode readers, same result. The installed driver comes from Microsoft (so it's the default one I'd say) and the device is enumerated as an "HID Keyboard Device" under the Windows Device Manager.

Any suggestions?