node-hid/node-hid #387: HID-hidraw.node no such file or directory

riccardocavallin, 2020-08-12:
I'm trying to install an application that requires this module.
When I launch 'yarn install' this error comes up: 'cp: cannot stat 'node_modules/node-hid/build/Release/HID-hidraw.node': No such file or directory'.
I've tried also 'yarn add node-hid' with the same result.
The operating system is macOS 10.15.6 and I'm running node v14.5.0, yarn 1.22.4 and npm 6.14.5,

todbot, 2020-08-12:
Could you provide a link to the project in question? Or a simplified project that recreates the error you're seeing?  I've not seen this error before, and normally `HID-hidraw.node` is only built for Linux, not MacOS.  And why aren't you getting a prebuilt binary?  

I just tested `node-hid` with Node v14 using npm on a fresh project and it installs and runs correctly.  The exact steps I did were:
```
nvm use v14
mkdir node-hid-test-nodev14 && cd !$
npm init -y
npm install node-hid --save
node ./node_modules/node-hid/src/show-devices.js
```

riccardocavallin, 2020-08-12:
This is the [project](https://github.com/radixdlt/desktop-wallet) and I'm just following the instructions in the readme.
Also I don't think the module is needed to run the project, but it does not start unless the problem is fixed.

todbot, 2020-08-12:
Ah, I see. It's a problem with `desktop-wallet`'s install process. It is making assumptions about how to install the native parts of modules.  Check out the "addbindings" script:
https://github.com/radixdlt/desktop-wallet/blob/release/2.0.0-beta.3/package.json#L32

As a work-around, you can change "HID-hidraw.node" to "HID.node".  

And then file an issue with `desktop-wallet`. This is not a node-hid issue.

closed 2020-08-12 by todbot

riccardocavallin, 2020-08-13:
Ok thank you.