node-hid/node-hid #391: could not read/write on Windows

MichaelZou1, 2020-08-25:
**environment**: Win10

**code operations**:
1.  open device thru vid and pid
2. print device info
3. try to read/write 

**problem**:
no matter reading or writing operation, an error was thrown  in step3, it's ok for  step2 to print device info.
`Error: Cannot write to HID device
    at Timeout._onTimeout (C:\Users\4paws\test\test.js:23:13)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)`

**following is the code snippet**:
```
var HID = require('node-hid');
var device = new HID.HID(5426, 552);
console.log(device)

var buttonGetDataCmd = [ 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02 ];

device.on('data', function(data) {
    var msg = '';
    var buttonState = data[0];
    if( buttonState === 0x17 ) {
        msg = "Lid opened!";
    } else if( buttonState === 0x16 ) {
        msg = "Button pushed!";
    } else if( buttonState === 0x15 ) {
        msg = "waiting...";
    }
    console.log('button data', data, msg);
});

setInterval( function() {
    device.write( buttonGetDataCmd );
}, 100)```

anybody has some problem?

todbot, 2020-08-25:
Hi,
Without knowing the HID reports (provided by the device in the HID Report Descriptor), it's impossible to tell if your code is correct or not.

However, from that code perhaps your device has two 8-byte reports, one IN and one OUT, using no report Ids.  If that is the case, then your `buttonGetDataCmd` array should be prepended with a zero byte (where the reportId would normally go), e.g.:
```js
var buttonGetDataCmd = [ 0x00, 0x08, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x02 ];
```
Also you probably want to add another timeout to you code so it doesn't exit before the response is received. For example, at the very bottom of your program add something like:
```js
setInterval( function() {  console.log("all done, goodbye"); }, 5000 );
```

MichaelZou1, 2020-08-26:
### Background

The HID device is "Razer BlackWidow Elite" keyboard. I'd like control the led light like Razer Synapse. it's the first time for me to look into USB functions, and the code snippet of original post just copy  **test-bigredbutton.js** with only vid,pid modified.

### HID Report Descriptor

"Razer BlackWidow Elite"  has three interfaces

 ```shell
interface0: boot keyboard

interface1: keyboard
interface1 HID report discriptor:05010906a1018501050719e029e71500250175019508810219002aff00150026ff007508950e8100c0050c0901a101850219002a3c021500263c029501751081007508950d8101c005010980a10185031981298315002501750195038102950581017508950e8101c005010900a10185040903150026ff00350046ff007508950f8100c005010900a10185050903150026ff00350046ff007508950f8100c0

interface2: Mouse(Usage 01)
interface2 HID report discriptor:05010902a1010901a100050919012905150025017501950581027501950381030600ff0940750895021581257f8102050109381581257f7508950181060930093116008026ff7f751095028106c00600ff0902150025017508955ab101c0
 ```

The descriptor parser shows HID descriptor only defines input report format.  So I used wireshard to capture out report data while changing  led settings.
The wireshark log shows feature report was sent using interface 2 and report id 0.

### current code

```javascript
var HID = require('node-hid');

var devices = HID.devices();
var deviceInfo = devices.find( function(d) {
    var israzer = d.vendorId===5426 && d.productId==552;
    return israzer && d.usagePage===1 && d.usage===2;
});

var device;
if( deviceInfo ) {
  device = new HID.HID( deviceInfo.path );
  // ... use device
  console.log(device, device.getDeviceInfo())
} else {
  console.log('could not find device')
}

function hexToBytes(hex) {
    for (var bytes = [], c = 0; c < hex.length; c += 2)
    bytes.push(parseInt(hex.substr(c, 2), 16));
    return bytes;
}

var report = hexToBytes('001f000000060f0201050401280000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000002200')

report.unshift(0)
setInterval( function() {
      device.sendFeatureReport( report);
}, 100);

setInterval( function() {  console.log("all done, goodbye"); }, 5000 );
```

it works :-)

closed 2020-08-26 by MichaelZou1