node-hid/node-hid #392: Node-hid cannot be installed with electron10

Sergey-Churilin, 2020-08-28:
I guess that the problem is in rebuilding modules.

To reproduce, please download sample and run npm install https://github.com/Sergey-Churilin/Electron-samples/tree/node-hid-electron10

todbot, 2020-08-28:
Can you provide the error message you are seeing?

todbot, 2020-08-28:
Your app had several bugs in it, not related to `node-hid`.  I've posted a fixed version here:
https://github.com/todbot/Electron-samples/tree/node-hid-electron10

Sergey-Churilin, 2020-08-31:
Hi @todbot , thank you very much for a quick look and answer. I'm still observing the problem during npm install event with your example. 
If I install manually separately node-hid, it goes smoothly. With full setup it fails.
1) If you run npm run build, will node-hid be rebuilt without issues? It fails for me.
2) Maybe you know what can be the root cause? Here is the partial stack of initial full installation

> 
1676 verbose lifecycle node-hid@1.3.0~install: CWD: C:\Users\User\Downloads\Electron-samples-node-hid-electron10\app\node_modules\node-hid
1677 silly lifecycle node-hid@1.3.0~install: Args: [ '/d /s /c', 'prebuild-install || node-gyp rebuild' ]
1678 timing audit submit Completed in 899ms
1679 http fetch POST 200 https://registry.npmjs.org/-/npm/v1/security/audits/quick 899ms
1680 timing audit body Completed in 0ms
1681 silly lifecycle node-hid@1.3.0~install: Returned: code: 1  signal: null
1682 info lifecycle node-hid@1.3.0~install: Failed to exec install script
1683 timing action:install Completed in 6081ms
1684 verbose unlock done using C:\Users\User\AppData\Roaming\npm-cache\_locks\staging-046e6bfcc9312db9.lock for C:\Users\User\Downloads\Electron-samples-node-hid-electron10\app\node_modules\.staging
1685 timing stage:rollbackFailedOptional Completed in 95ms
1686 timing stage:runTopLevelLifecycles Completed in 9320ms
...
1690 verbose stack Error: node-hid@1.3.0 install: `prebuild-install || node-gyp rebuild`
1690 verbose stack Exit status 1
1690 verbose stack     at EventEmitter.<anonymous> (C:\Users\User\AppData\Roaming\npm\node_modules\npm\node_modules\npm-lifecycle\index.js:332:16)
1690 verbose stack     at EventEmitter.emit (events.js:315:20)
1690 verbose stack     at ChildProcess.<anonymous> (C:\Users\User\AppData\Roaming\npm\node_modules\npm\node_modules\npm-lifecycle\lib\spawn.js:55:14)
1690 verbose stack     at ChildProcess.emit (events.js:315:20)
1690 verbose stack     at maybeClose (internal/child_process.js:1021:16)
1690 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
1691 verbose pkgid node-hid@1.3.0
1692 verbose cwd C:\Users\User\Downloads\Electron-samples-node-hid-electron10\app
1693 verbose Windows_NT 10.0.18363
1694 verbose argv "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\User\\AppData\\Roaming\\npm\\node_modules\\npm\\bin\\npm-cli.js" "install" "--production" "--cache-min" "999999999"
1695 verbose node v12.18.3
1696 verbose npm  v6.14.8
1697 error code ELIFECYCLE
1698 error errno 1
1699 error node-hid@1.3.0 install: `prebuild-install || node-gyp rebuild`
1699 error Exit status 1
1700 error Failed at the node-hid@1.3.0 install script.
1700 error This is probably not a problem with npm. There is likely additional logging output above.
1701 verbose exit [ 1, true ]

todbot, 2020-09-07:
Your setup sounds like it has some issues.  On Windows 10 x64, `npm install` on my fork installs without compiling, using a prebuilt binary.

todbot, 2020-09-07:
If for some reason you're triggering a rebuild, It looks like you're using a non-compliant compiler?  Compilation requirements are described here: https://github.com/node-hid/node-hid#windows-7-8-10
These are not node-hid's requirements, but node-gyp's requirements: https://github.com/nodejs/node-gyp

Sergey-Churilin, 2020-09-08:
Hi @todbot , thank you very much for looking into my issue, I didn't change my setup for a long time, the issue began to occur when I started migration to electron 10. Last working electron version was 9.x.x

todbot, 2020-09-09:
Apologies, I didn't read carefully that you were on Electron v10.
`node-hid` is not yet supported on Electron v10. 
But I'm working on it right now.

todbot, 2020-09-09:
If you really want to use node-hid with Electron v10, you can copy what is done here:
  https://github.com/todbot/electron-hid-test/blob/master/package.json

The relevant changes in `package.json` are:
```
  "devDependencies": {
    "electron": "^10.1.1",
    "electron-rebuild": "electron/electron-rebuild#malept/fix/actually-run-node-gyp-rebuild"
  },
  "dependencies": {
    "node-hid": "github:node-hid/node-hid"
  }
```

Note that a new branch of `electron-rebuild` is required because Electron v10 broke `electron-rebuild`. Also note that the github version of `node-hid` is required because Electron v10 broke `node-abi`.

In general, if you are using native modules, do not use the latest version of Electron or Nodejs.

Sergey-Churilin, 2020-09-10:
Thank you very much for looking into that again, really appreciate it!

Sergey-Churilin, 2020-09-16:
Hi @todbot , did you have a chance to open an issue on Electron itself, as you said that
- Electron v10 broke electron-rebuild
- Electron v10 broke node-abi

todbot, 2020-09-16:
Hi @Sergey-Churilin,
Issues had already been opened on the issue.  The `electron-rebuild` issue as recently incorporated into a release, see https://github.com/electron/electron-rebuild/issues/404 
As for `node-abi` (the larger issue wrt ABI detection), the issue for that is here: https://github.com/lgeiger/node-abi/issues/90 and I've added a test PR here: https://github.com/lgeiger/node-abi/pull/94

Draginfable, 2020-11-19:
Hello @todbot,
I tried to test the latest version that you released but it seems that you haven't published it to the npm registry yet.

Draginfable, 2020-11-20:
After testing everything seems to be working fine.

lukebayes, 2020-11-22:
Hey @todbot,

First, thank you so much for all your hard work on this library!

I'm attempting to use `node-hid` in the context of an Electron application on a Raspberry Pi (and hopefully Ubuntu 20.04 workstation(s)).

I've managed to get everything up to the latest versions as shown in your Electron sample project.

My runtime versions are as follows:
```
Launching Electron with:
{
  node: '12.18.3',
  v8: '8.7.220.25-electron.0',
  uv: '1.38.0',
  zlib: '1.2.11',
  brotli: '1.0.7',
  ares: '1.16.0',
  modules: '85',
  nghttp2: '1.41.0',
  napi: '6',
  llhttp: '2.0.4',
  http_parser: '2.9.3',
  openssl: '1.1.1',
  icu: '67.1',
  unicode: '13.0',
  electron: '11.0.2',
  chrome: '87.0.4280.67'
}
````
I'm using `node-hid@1.3.2`, and `node-abi@2.19.3`.

The good news is, that your sample project does load correctly and list devices, the bad news is that when I attempt to open a device (e.g., `const dev = hid.HID(devicePath)`), I get the dreaded:

```
Renderer process crashed - see https://www.electronjs.org/docs/tutorial/application-debugging for potential debugging information.
```
FWIW, I did have issues getting the HID devices to be accessible to a non-root user, but was able to finally get a udev rule that worked, so I believe I've got appropriate access to the devices.

I have also tried rolling back Electron to release version 9.3.4 (and then run `electron-rebuild`), but still get essentially the same behavior (seems like the language the error changed slightly since then).

Any tips or pointers would be greatly appreciated.

Thanks again!

Luke

lukebayes, 2020-11-22:
Quick update.

I was able to roll Electron back to version 8.5.5, and things seem to be working there.

todbot, 2020-11-24:
node-hid has recently been rewritten to use N-API instead of NAN, with node-hid@2.0.0-0.
This change makes prebuilds much simpler and does not rely on node-abi and tracking Node/Electron ABI versions.

closed 2020-11-24 by todbot

lukebayes, 2020-11-24:
The fix is confirmed to resolve the issue. Thanks, and great timing!