node-hid/node-hid #393: Cross-compiling: node-gyp unconditionally executes pkg-config on Linux

eepp, 2020-08-31:
Currently, node-hid's `binding.gyp` contains this:

```
                [ 'OS=="linux"', {
                    'conditions': [
                        [ 'driver=="libusb"', {
                            'sources': [ 'hidapi/libusb/hid.c' ],
                            'include_dirs+': ['<!@(pkg-config libusb-1.0 --cflags-only-I | sed s/-I//g)']
                        }],
                        [ 'driver=="hidraw"', {
                            'sources': [ 'hidapi/linux/hid.c' ]
                        }]
                    ]
                }],
```

This means, on Linux: add what `pkg-config libusb-1.0 --cflags-only-I | sed s/-I//g` prints to the list of header directories.

However, when cross-compiling, it makes little sense to use pkg-config. If I configure some project X to be installed under `/usr` on the target (passing `--prefix=/usr`), then typically that project's generated `.pc` file will indicate `/usr/include` for headers and `/usr/lib` for libraries. However, when cross-compiling some project Y which depends on X's headers and libraries, they won't be found in `/usr/include` and `/usr/lib`.

In this case, project X is libusb and project Y is node-hid.

The way to propagate header and library directories when cross-compiling is to use the good old `CFLAGS` and `LDFLAGS` environment variables. And GYP does take those into account (or, at least, the Makefiles it generates).

So there's no need to use pkg-config when cross-compiling.

The current issue is that, if the system with which you build node-hid does _not_ have libusb installed, then pkg-config doesn't find any corresponding `.pc` file, doesn't exit with status 0, and the whole node-gyp operation fails.

Is there a way to conditionally _not_ use pkg-config when building node-hid, for example with a specific environment variable?

What I do currently is I maintain a fork of node-gyp with a patch removing `binding.gyp`'s `pkg-config`. This complicates the build, however. I'd like to be able to just specify a `node-gyp` dependency in my project's `package.json` and use the npm public registry.

todbot, 2020-08-31:
Hi @eep,
This is an interesting problem.  While I am not a GYP expert (it can be a deep and confusing tool), I think perhaps the following change to https://github.com/node-hid/node-hid/blob/ed6ebcc0e7ab700fc365d32b4e01b88f91d4f26f/binding.gyp#L74 may do what you're asking:
```
     'include_dirs+': ['<!@(echo ${CFLAGS:-`pkg-config libusb-1.0 --cflags-only-I`} | sed s/-I//g)']
```
This lets you override the calling of pkg-config by setting `CFLAGS`.

eepp, 2020-08-31:
Thank you for taking the time to answer.

I'm afraid the solution you suggest is not ideal, for the following reasons:

* `CFLAGS` can contain other things than include directories: any C compiler flag.
* `CXXFLAGS` can also contain include directories.
* `CPPFLAGS` can also contain include directories.

In other words, having a `CFLAGS` environment variable is independent from using pkg-config or not.

See #394 for an alternative.

todbot, 2020-08-31:
This is perfect. The more I was sitting with that CFLAGS hack the less I liked it.  I'll merge your PR presently and update the README. Thanks!

eepp, 2020-08-31:
Thank you @todbot!

eepp, 2020-08-31:
BTW, are you planning to make a patch release including this soon?

I'd really like to just specify `"node-hid": "^1.3.1"` as a dependency in my `package.json` instead of this Git repository.

todbot, 2020-09-10:
Hi! Sorry about the delay. It's in the main branch in the repo.  I've not published it yet as I'm waiting another day or so for `electron-rebuild` devs to fix it for Electron v10. (There's an approved PR that I've verified works for node-hid. Hopefully it'll get merged & published today)

In any case, expect 1.3.1 to be on npm by this weekend.