node-hid/node-hid #396: FreeBSD 12.1 Undefined symbol "hid_init"

ruslan-gennadievich, 2020-09-07:
Hello
I builded hidapi via
https://github.com/libusb/hidapi/blob/master/.builds/freebsd.yml

after am try use node-hid, but have error:
ld-elf.so.1: /usr/hsm/node_modules/node-hid/build/Release/HID.node: Undefined symbol "hid_init"

How i can use node-hid on freebsd12.1?

todbot, 2020-09-08:
Node-hid has not been tested in FreeBSD, but it looks like it just cannot find the `hidapi` library.  And it uses its own copy of `hidapi` so you shouldn't have to install your own (It looks like that hidapi installer you link installs to `$DESTDIR` instead of a system library directory anyway)

Could you provide the output you see when doing `npm install node-hid`

ruslan-gennadievich, 2020-09-09:
> node-hid@1.3.0 install /usr/hsm/node_modules/node-hid
> prebuild-install || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=14.8.0 runtime=node arch=x64 libc= platform=freebsd)
gmake: Entering directory '/usr/hsm/node_modules/node-hid/build'
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
c++: warning: argument unused during compilation: '-e xceptions' [-Wunused-command-line-argument]
c++: warning: argument unused during compilation: '-e xceptions' [-Wunused-command-line-argument]
  SOLINK_MODULE(target) Release/obj.target/HID.node
  COPY Release/HID.node
gmake: Leaving directory '/usr/hsm/node_modules/node-hid/build'
___________
No errors. But if i run my programm, i see:
ld-elf.so.1: /usr/hsm/node_modules/node-hid/build/Release/HID.node: Undefined symbol "hid_init"

ruslan-gennadievich, 2020-09-09:
I think that binding.gyp must have freebsd section

todbot, 2020-09-09:
I think you are correct. I just got a FreeBSD 12 system up to test on, and I can try some experiments. But if you have experience with FreeBSD compilation and have ideas about what should go in the binding.gyp, I'd very much like to hear them.

todbot, 2020-09-09:
Okay I think I may have a fix checked in.  Can you try it?

ruslan-gennadievich, 2020-09-09:
[ 'OS=="freebsd"', {
                  'libraries': ['-lusb']
                }], # OS==freebsd

This can't find libraries. But it work if write full path:
                [ 'OS=="freebsd"', {
    'libraries': ['/usr/local/lib/libhidapi.so']
                }], # OS==freebsd

todbot, 2020-09-09:
I do not understand, hidapi is built locally, not in /usr/local.  Are you referring to /usr/lib/libusbhid?

todbot, 2020-09-09:
Perhaps you are not running `npm run prepublishOnly`?

ruslan-gennadievich, 2020-09-09:
hidapi that built inside in node-hid, mabye not correct, because freebsd have specific libusb.

I clone hidapi from git and compile it. This lib install to /usr/local/lib/libhidapi.so
After i recompile node-hid with 'libraries': ['/usr/local/lib/libhidapi.so'] in binding.gyp

todbot, 2020-09-10:
The hidapi submodule builds against FreeBSD libusb.  And it is a more recent version of hidapi than is in the `pkg` database.   Can you please try the following and tell me if it works:

```
git clone https://github.com/node-hid/node-hid
cd node-hid
npm run prepublishOnly
npm install
sudo npm run showdevices
```

ruslan-gennadievich, 2020-09-10:
root@OPNsense:/usr # cd node-hid
root@OPNsense:/usr/node-hid # npm run prepublishOnly
npm WARN npm npm does not support Node.js v14.8.0
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 6, 8, 9, 10, 11, 12, 13.
npm WARN npm You can find the latest version at https://nodejs.org/

> node-hid@1.3.1 prepublishOnly /usr/node-hid
> git submodule update --init

Submodule 'hidapi' (https://github.com/libusb/hidapi.git) registered for path 'hidapi'
Cloning into '/usr/node-hid/hidapi'...
Submodule path 'hidapi': checked out 'ca1a2d6efae8d372587f4c13f60632916681d408'
root@OPNsense:/usr/node-hid # npm install
npm WARN npm npm does not support Node.js v14.8.0
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 6, 8, 9, 10, 11, 12, 13.
npm WARN npm You can find the latest version at https://nodejs.org/
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN lifecycle node-hid@1.3.1~install: cannot run in wd node-hid@1.3.1 prebuild-install || node-gyp rebuild (wd=/usr/node-hid)
npm notice created a lockfile as package-lock.json. You should commit this file.
added 238 packages from 172 contributors and audited 239 packages in 7.936s
found 0 vulnerabilities

root@OPNsense:/usr/node-hid # npm run showdevices
npm WARN npm npm does not support Node.js v14.8.0
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 6, 8, 9, 10, 11, 12, 13.
npm WARN npm You can find the latest version at https://nodejs.org/

> node-hid@1.3.1 showdevices /usr/node-hid
> node src/show-devices.js

/usr/node-hid/node_modules/bindings/bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 → /usr/node-hid/build/HID.node
 → /usr/node-hid/build/Debug/HID.node
 → /usr/node-hid/build/Release/HID.node
 → /usr/node-hid/out/Debug/HID.node
 → /usr/node-hid/Debug/HID.node
 → /usr/node-hid/out/Release/HID.node
 → /usr/node-hid/Release/HID.node
 → /usr/node-hid/build/default/HID.node
 → /usr/node-hid/compiled/14.8.0/freebsd/x64/HID.node
 → /usr/node-hid/addon-build/release/install-root/HID.node
 → /usr/node-hid/addon-build/debug/install-root/HID.node
 → /usr/node-hid/addon-build/default/install-root/HID.node
 → /usr/node-hid/lib/binding/node-v83-freebsd-x64/HID.node
    at bindings (/usr/node-hid/node_modules/bindings/bindings.js:126:9)
    at loadBinding (/usr/node-hid/nodehid.js:25:42)
    at Object.showdevices [as devices] (/usr/node-hid/nodehid.js:127:5)
    at Object.<anonymous> (/usr/node-hid/src/show-devices.js:13:29)
    at Module._compile (internal/modules/cjs/loader.js:1251:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1272:10)
    at Module.load (internal/modules/cjs/loader.js:1100:32)
    at Function.Module._load (internal/modules/cjs/loader.js:962:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47 {
  tries: [
    '/usr/node-hid/build/HID.node',
    '/usr/node-hid/build/Debug/HID.node',
    '/usr/node-hid/build/Release/HID.node',
    '/usr/node-hid/out/Debug/HID.node',
    '/usr/node-hid/Debug/HID.node',
    '/usr/node-hid/out/Release/HID.node',
    '/usr/node-hid/Release/HID.node',
    '/usr/node-hid/build/default/HID.node',
    '/usr/node-hid/compiled/14.8.0/freebsd/x64/HID.node',
    '/usr/node-hid/addon-build/release/install-root/HID.node',
    '/usr/node-hid/addon-build/debug/install-root/HID.node',
    '/usr/node-hid/addon-build/default/install-root/HID.node',
    '/usr/node-hid/lib/binding/node-v83-freebsd-x64/HID.node'
  ]
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! node-hid@1.3.1 showdevices: `node src/show-devices.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the node-hid@1.3.1 showdevices script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

todbot, 2020-09-10:
I do not understand then. The exact same sequence of commands run on a fresh FreeBSD v12 install works.  Here's what I see when I run `npm install`:
```
[todbot@freebsd12 ~/projects/node-hid]$ npm install
npm WARN npm npm does not support Node.js v14.4.0
npm WARN npm You should probably upgrade to a newer version of node as we
npm WARN npm can't make any promises that npm will work with this version.
npm WARN npm Supported releases of Node.js are the latest release of 6, 8, 9, 10, 11, 12, 13.
npm WARN npm You can find the latest version at https://nodejs.org/
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported

> node-hid@1.3.1 install /usr/home/todbot/projects/node-hid
> prebuild-install || node-gyp rebuild

gmake: Entering directory '/usr/home/todbot/projects/node-hid/build'
  CC(target) Release/obj.target/hidapi/hidapi/libusb/hid.o
  AR(target) Release/obj.target/hidapi.a
  COPY Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
c++: warning: argument unused during compilation: '-e xceptions' [-Wunused-command-line-argument]
c++: warning: argument unused during compilation: '-e xceptions' [-Wunused-command-line-argument]
  SOLINK_MODULE(target) Release/obj.target/HID.node
  COPY Release/HID.node
gmake: Leaving directory '/usr/home/todbot/projects/node-hid/build'
npm notice created a lockfile as package-lock.json. You should commit this file.
added 238 packages from 172 contributors and audited 239 packages in 13.929s
found 0 vulnerabilities
```

todbot, 2020-09-10:
Beforehand did you `pkg install gmake gcc`?

ruslan-gennadievich, 2020-09-10:
> Beforehand did you `pkg install gmake gcc python`?

Yes, a install it.

tod, maybe you can connect to my freebsd for testing?
give me your email, and i send ssh password from my test-server

ruslan-gennadievich, 2020-09-11:
Im check, now correct build and work in FreeBSD.
Thanks for fix!

closed 2020-09-16 by ruslan-gennadievich