node-hid/node-hid #397: Issues in prepublishOnly script

bipinmehta, 2020-09-10:
As per the instructions in the repository, I installed the package and run the commands where I got following errors:
1. npm run prepublishOnly =>Issues reported => ERR fatal: not a git repository
     Exit status: 128

2. node ./src/show-devices.js => Issue reported => Could not locate the bindings file. Tried: .... names of paths tried
   in /node_modules/bindings/bindings.js:135

Resolutions to these errors will be highly appreciated. Regards

todbot, 2020-09-10:
Hi,
Please provide OS type & version, and Node and npm version. Also please provide exact commands typed and exact screen output.

For instance, this is what I see on MacOS 10.15.6, Node v14.8.0 and npm 6.14.7:
```
% git clone https://github.com/node-hid/node-hid.git
Cloning into 'node-hid'...
remote: Enumerating objects: 122, done.
remote: Counting objects: 100% (122/122), done.
remote: Compressing objects: 100% (81/81), done.
remote: Total 1379 (delta 76), reused 79 (delta 41), pack-reused 1257
Receiving objects: 100% (1379/1379), 371.86 KiB | 1.80 MiB/s, done.
Resolving deltas: 100% (845/845), done.

% cd node-hid

% npm run prepublishOnly

> node-hid@1.3.1 prepublishOnly /Users/tod/tmp/node-hid-397/node-hid
> git submodule update --init

Submodule 'hidapi' (https://github.com/libusb/hidapi.git) registered for path 'hidapi'
Cloning into '/Users/tod/tmp/node-hid-397/node-hid/hidapi'...
Submodule path 'hidapi': checked out 'ca1a2d6efae8d372587f4c13f60632916681d408'

% npm install --build-from-source
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated har-validator@5.1.5: this library is no longer supported

> node-hid@1.3.1 install /Users/tod/tmp/node-hid-397/node-hid
> prebuild-install || node-gyp rebuild

  CC(target) Release/obj.target/hidapi/hidapi/mac/hid.o
  LIBTOOL-STATIC Release/hidapi.a
  CXX(target) Release/obj.target/HID/src/HID.o
  SOLINK_MODULE(target) Release/HID.node
npm notice created a lockfile as package-lock.json. You should commit this file.
added 238 packages from 172 contributors and audited 239 packages in 11.075s

4 packages are looking for funding
  run `npm fund` for details

found 0 vulnerabilities
```

bipinmehta, 2020-09-10:
Sir,
Thank you for the quick response. Please find the details below:
Versions: Mac OS Catalina 10.15.6; Node: 12.14.0; npm: 6.14.8

Exact command typed:
1.  npm run prepublishOnly
 Response:
> node-hid@1.3.0 prepublishOnly
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app
> git submodule update --init

fatal: not a git repository (or any of the parent directories): .git
npm ERR! code ELIFECYCLE
npm ERR! errno 128
npm ERR! node-hid@1.3.0 prepublishOnly: `git submodule update --init`
npm ERR! Exit status 128
npm ERR!
npm ERR! Failed at the node-hid@1.3.0 prepublishOnly script.
npm ERR! This is probably not a problem with npm. There is likely
additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/bipin289/.npm/_logs/2020-09-10T14_55_26_378Z-debug.log
bipin289@Bipins-MacBook-Pro my-hmi-app %

2. node ./src/show-devices.js
  Response:
bipin289@Bipins-MacBook-Pro my-hmi-app % node ./src/show-devices.js
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/node_modules/bindings/bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/Debug/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/Release/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/out/Debug/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/Debug/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/out/Release/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/Release/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/default/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/compiled/12.14.0/darwin/x64/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/release/install-root/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/debug/install-root/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/default/install-root/HID.node
 →
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/lib/binding/node-v72-darwin-x64/HID.node
    at bindings
(/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/node_modules/bindings/bindings.js:126:9)
    at loadBinding
(/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/nodehid.js:23:36)
    at Object.showdevices [as devices]
(/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/nodehid.js:114:3)
    at Object.<anonymous>
(/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/src/show-devices.js:13:29)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1047:10)
    at internal/main/run_main_module.js:17:11 {
  tries: [

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/Debug/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/Release/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/out/Debug/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/Debug/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/out/Release/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/Release/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/default/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/compiled/12.14.0/darwin/x64/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/release/install-root/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/debug/install-root/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/default/install-root/HID.node',

'/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/lib/binding/node-v72-darwin-x64/HID.node'
  ]
}
                --------------------------------

Awaiting guidance. Regards

Bipin Mehta

bipinmehta, 2020-09-10:
Sir,Thank you for the quick response. Please find the details below:Versions: Mac OS Catalina 10.15.6; Node: 12.14.0; npm: 6.14.8
Exact command typed:1.  npm run prepublishOnly Response:> node-hid@1.3.0 prepublishOnly /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app
> git submodule update --init

fatal: not a git repository (or any of the parent directories): .git
npm ERR! code ELIFECYCLE
npm ERR! errno 128
npm ERR! node-hid@1.3.0 prepublishOnly: `git submodule update --init`
npm ERR! Exit status 128
npm ERR! 
npm ERR! Failed at the node-hid@1.3.0 prepublishOnly script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /Users/bipin289/.npm/_logs/2020-09-10T14_55_26_378Z-debug.log
bipin289@Bipins-MacBook-Pro my-hmi-app %       

2. node ./src/show-devices.js  Response:bipin289@Bipins-MacBook-Pro my-hmi-app % node ./src/show-devices.js
/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/node_modules/bindings/bindings.js:135
  throw err;
  ^

Error: Could not locate the bindings file. Tried:
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/Debug/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/Release/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/out/Debug/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/Debug/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/out/Release/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/Release/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/default/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/compiled/12.14.0/darwin/x64/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/release/install-root/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/debug/install-root/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/default/install-root/HID.node
 → /Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/lib/binding/node-v72-darwin-x64/HID.node
    at bindings (/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/node_modules/bindings/bindings.js:126:9)
    at loadBinding (/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/nodehid.js:23:36)
    at Object.showdevices [as devices] (/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/nodehid.js:114:3)
    at Object.<anonymous> (/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/src/show-devices.js:13:29)
    at Module._compile (internal/modules/cjs/loader.js:959:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:995:10)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Function.Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module.runMain (internal/modules/cjs/loader.js:1047:10)
    at internal/main/run_main_module.js:17:11 {
  tries: [
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/Debug/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/Release/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/out/Debug/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/Debug/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/out/Release/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/Release/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/build/default/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/compiled/12.14.0/darwin/x64/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/release/install-root/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/debug/install-root/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/addon-build/default/install-root/HID.node',
    '/Users/bipin289/Desktop/Working-Dir/nodejs-usb-projects/my-hmi-app/lib/binding/node-v72-darwin-x64/HID.node'
  ]
}
                --------------------------------

Awaiting guidance. Regards
Bipin Mehta

todbot, 2020-09-10:
I did not see how you originally obtained this repository?  What commands did you use to obtain it?

bipinmehta, 2020-09-10:
I worked as below:

Opened your repository
Downloaded the zip file in to my dir: my-hmi-app
Unzipped the file
Opened in VSCode
Followed steps given in README.md
Compiling from source

Mac OS X 10.8+

   - Xcode <https://itunes.apple.com/us/app/xcode/id497799835?mt=12>.
   (already installed ver 11.7)

<https://github.com/node-hid/node-hid#windows-7-8-10>Building node-hid from
source, for your projects

npm install node-hid --build-from-source

<https://github.com/node-hid/node-hid#build-node-hid-for-node-hid-development>
Build node-hid for node-hid development

   - check out a copy of this repo
   - change into its directory
   - update the submodules
   - build the node package

For example:

git clone https://github.com/node-hid/node-hid.git
cd node-hid                                        # must change into
node-hid directory
npm install -g rimraf                              # just so it
doesn't get 'clean'ed
npm run prepublishOnly                             # get the needed
hidapi submodule
npm install --build-from-source                    # rebuilds the
module with C code
node ./src/show-devices.js

You may see some warnings from the C compiler as it compiles hidapi
<https://github.com/libusb/hidapi> (the underlying C library node-hid uses).
This is expected.

---------------------------------------------------------

Results I obtained for npm run prepublishOnly and node
./src/show-devices.js are already given to you in my earlier mail.

Awaiting further guidance. Regards

Bipin Mehta

todbot, 2020-09-10:
Ahh that is the problem. You cannot run `npm run prepublishOnly` from a .zip download.  You should do `git clone https://github.com/node-hid/node-hid`

If you must use a zip download, then can do the following. 
1. Go to  repo and download zip
2. Go to https://github.com/libusb/hidapi/tree/ca1a2d6efae8d372587f4c13f60632916681d408 and download hidapi submodule zip
3. Unzip node-hid zip
4. Inside of resulting node-hid directory, unzip the hidapi zip and rename it 'hidapi'
5. Run npm install

Why are you using node-hid from a zip?  It is highly irregular. Most users just do `npm install node-hid`

closed 2020-09-10 by todbot

bipinmehta, 2020-09-11:
Sir,
Thank you for your valuable guidance. I have cloned directly and installed in root directory. Can I relocate this folder to elsewhere in my project dir.? Regards

todbot, 2020-09-11:
What is it you are trying to do?  Normally one does not deal with the source directories of node modules directly.

bipinmehta, 2020-09-14:
Thanks for the support.