node-hid/node-hid #398: missing prebuilds of current Node.js LTS (12.18.3)

gre, 2020-09-11:
- according to https://nodejs.org/en/ the current version LTS is 12.18.3
- https://nodejs.org/en/download/releases/ says that this version has node module version v72
- https://github.com/node-hid/node-hid/releases/tag/v1.3.1 no longer have v72 in the list of prebuilds packages (in contrary to 1.3.0).

k-matsuzawa, 2020-09-11:
Please refer to the following:
https://github.com/lgeiger/node-abi/issues/90

If you are in a hurry to release, it is recommended to temporarily fix node-abi to v2.18.0.

todbot, 2020-09-11:
Yes please use node-hid@1.3.0 until node-abi gets fixed

gre, 2020-11-23:
I've been trying latest `node-hid@1.3.2` in combination of electron-builder 22.8.0 but still on Electron 9 and I get

```
I.e: dyld: lazy symbol binding failed: Symbol not found: __ZN2v816FunctionTemplate3NewEPNS_7IsolateEPFvRKNS_20FunctionCallbackInfoINS_5ValueEEEENS_5LocalIS4_EENSA_INS_9SignatureEEEiNS_19ConstructorBehaviorENS_14SideEffectTypeE
  Referenced from: ./node_modules/node-hid/build/Release/HID.node
  Expected in: flat namespace
```

I'm not sure if node-hid@1.3.2 expects Electron 10, but it seems to fail with v9. (I didn't try yet on v10)

todbot, 2020-11-23:
Hi two questions:
1. What OS are you on?
2. Are you doing `electron-rebuild` to get the Electron build of `node-hid`

gre, 2020-11-23:
1. MacOS Catalina (10.15.7)
2. Mmh good point, we might have a cache in place that didn't recall a rebuild, let me run it again to be sure.

todbot, 2020-11-23:
I just verified that when running `https://github.com/todbot/electron-hid-test/` modified to use Electron v9.3.4 that it fails.  It's likely I cannot fix this as it's likely a problem with `node-abi` still, but I'll look into it.

gre, 2020-11-23:
ok, thanks a lot. I tried a full clear and unfortunately am still falling into the issue indeed, hope we figure it out 🤞 
tell me if you want me to test against Electron v10 (not sure if it's easy to do or if you already know that it does work)

Thanks!

todbot, 2020-11-23:
Yes, it looks like `node-abi` is reporting wrong ABI to `prebuid` still for many versions of Electron. Sigh.
Thankfully the PR for converting from NAN to NAPI looks to be working so a future version of `node-hid` won't have to deal with all the problems in `node-abi`. (I think it lost its main maintainer)

todbot, 2020-11-23:
Oh also, Electron v10's prebuild does appear to work.

todbot, 2020-11-24:
Hi @gre, could you try `node-hid@2.0.0-0` and let me know if the prebuilds work for you for any version of Electron you want?

gre, 2020-12-01:
Hi @todbot. I can make it work on Mac + Electron 9.
We'll unroll more QA in my team to validate it on other platform but so far so good.

Thanks a lot!

closed 2020-12-01 by gre