node-hid/node-hid #402: Rewrite using N-API?

Julusian, 2020-11-02:
Have you considered rewriting this library to use N-API (and https://github.com/nodejs/node-addon-api) instead of NAN?
N-API has abi stability, such that the same prebuilds will work with in all versions of node 10+, including with electron https://nodejs.org/api/n-api.html#n_api_n_api_version_matrix

The only downside I have seen to N-API is that it will require dropping support for node 8 and older, but as they are EOL and many other libraries have dropped support I do not see that as a problem.

If you are interested, I am happy to put the work in to port the code across

todbot, 2020-11-02:
Oh yes, I've totally considered it, but it's felt a pretty big undertaking on V8 C code I'm only peripherally (haha) familiar with.

If you're more familiar with NAN and N-API, I would very much appreciate the help.

todbot, 2020-11-02:
Also, I am curious how the prebuild system works with N-API and if it's at the mercy of `node-abi` the way all the other native modules are.

Julusian, 2020-11-02:
Thats reasonable, Ive already ported a couple of libraries to napi (https://github.com/julusian/node-freetype2 and working on https://github.com/julusian/node-jpeg-turbo), so I am happy with the process.

I don't think so, digging into prebuild it appears to use a library which at its heart gets the maximum supported napi version from `process.versions.napi`, and as node states that the versions will always be supported https://nodejs.org/en/docs/guides/abi-stability/#n-api it knows that all previous api versions will be supported.

Ill have a play, expect a PR soon

todbot, 2020-11-24:
Hi @Julusian, I've been testing this rewrite and it seems to work great.  I've merged the changes, set up prebuilds, and created a new release `node-hid@2.0.0-1`.

The switch to N-API has made the prebuild situation so much simpler for end users (same binary works for Node & Electron) and it lets us get away from tracking the latest ABIs, which `node-abi` has always had issues with.

Thank you!

todbot, 2020-12-10:
This rewrite looks like it's been working well.  Closing this issue. And thank you!

closed 2020-12-10 by todbot