node-hid/node-hid #404: Blink(1) led light not working with node-hid

mountaindude, 2020-11-18:
macOS 10.15.7 on Macbook Pro
Node.js v12.16.0

Blink(1) mk2 connected via USB hub.
Shows up in HID.devices():

```node.js
var HID = require('node-hid');
var devices = HID.devices();
console.log(devices);
```

```
...
...
  {
    vendorId: 10168,
    productId: 493,
    path: 'IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/blink(1) mk2@14300000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice',
    serialNumber: '........',
    manufacturer: 'ThingM',
    product: 'blink(1) mk2',
    release: 2,
    interface: 0,
    usagePage: 65280,
    usage: 1
  },
...
...
```

But when adding either 
```node.js
var device = new HID.HID('IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/blink(1) mk2@14300000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice');
```
or

```node.js
var device = new HID.HID(10168,493);
```

I get 

```
/Users/goran/code/node-hid-test1/node_modules/node-hid/nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with path IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/XHC1@14/XHC1@14000000/HS03@14300000/blink(1) mk2@14300000/IOUSBHostInterface@0/AppleUserUSBHostHIDDevice
    at new HID (/Users/goran/code/node-hid-test1/node_modules/node-hid/nodehid.js:49:17)
    at Object.<anonymous> (/Users/goran/code/node-hid-test1/index.js:9:14)
    at Module._compile (internal/modules/cjs/loader.js:1157:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1177:10)
    at Module.load (internal/modules/cjs/loader.js:1001:32)
    at Function.Module._load (internal/modules/cjs/loader.js:900:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:74:12)
    at internal/main/run_main_module.js:18:47
```

and 

```
/Users/goran/code/node-hid-test1/node_modules/node-hid/nodehid.js:49
    this._raw = new (Function.prototype.bind.apply(binding.HID,
                ^

Error: cannot open device with vendor id 0x27b8 and product id 0x1ed
    at new HID (/Users/goran/code/node-hid-test1/node_modules/node-hid/nodehid.js:49:17)
    at Object.<anonymous> (/Users/goran/code/node-hid-test1/index.js:12:14)
    at Module._compile (internal/modules/cjs/loader.js:1157:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1177:10)
    at Module.load (internal/modules/cjs/loader.js:1001:32)
    at Function.Module._load (internal/modules/cjs/loader.js:900:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:74:12)
    at internal/main/run_main_module.js:18:47
```

Not sure where to go from here, ideas welcome.

mountaindude, 2020-11-18:
Update: 

A reboot and reinstall of Xcode later those errors are gone. 
The sample node.js code for Blink(1) works - kind of. 

It outputs the expected text, but the leds don't blink or light up in any way.  
This is the same behaviour as I see in both older Blink1Control and newer Blink1Contrl2 apps.

Is hardware failure a possibility...? I've tried three Blink(1) so far. Might have to dig up the remaining ones too from storage and test them too. LED getting warm after many hours in use, solder joints loosening up, ...?
The microcontroller is working though (as it does report back to the controlling app), always something.

todbot, 2020-11-18:
It sounds like you may have very old mk2s where some could have a fault leading them to premature death. If it's getting warm I would not use it. Please send us an email at blink1 at thingm.com and we'll arrange replacements to be sent to you.

closed 2020-11-24 by todbot