node-hid/node-hid #405: Installing node-hid raspberry pi

RhinoDale, 2020-11-20:
I am using electron 9.3.4
raspian os

Here is the error
$ npm install node-hid --build-from-source

> node-hid@1.3.2 install /home/pi/checkthisrealty/node_modules/node-hid
> prebuild-install || node-gyp rebuild

/home/pi/.config/nvm/versions/node/v13.14.0/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin/node-gyp: 5: /home/pi/.config/nvm/versions/node/v13.14.0/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin/node-gyp: /usr/local/lib/node_modules/node-gyp/bin/node-gyp.js: not found
npm WARN The package jquery is included as both a dev and production dependency.

npm ERR! code ELIFECYCLE
npm ERR! syscall spawn
npm ERR! file sh
npm ERR! errno ENOENT
npm ERR! node-hid@1.3.2 install: `prebuild-install || node-gyp rebuild`
npm ERR! spawn ENOENT
npm ERR! 
npm ERR! Failed at the node-hid@1.3.2 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-11-20T21_36_26_835Z-debug.log

I think the problem is 
/usr/local/lib/node_modules/node-gyp/bin/node-gyp.js: not found
I do have
/usr/local/lib/node_modules/**npm**/node_modules/node-gyp/bin/node-gyp.js

But I do not know where to modify the script to use this directory.

Any assistance is welcomed.

todbot, 2020-11-20:
It looks like you're using `nvm` but your node installation is referencing `/usr/local/lib/node-modules`.  You should be using one or the other, so I think your node installation is partially broken.  On a Raspberry Pi OS system with nvm, normally doing `npm install node-hid` will work properly.

As a temporary work-around for your setup, you may be able to do `npm install -g node-gyp`

As a longer-term solution, I would recommend using `apt` to remove all Node-related packages and rely exclusively on `nvm`.

RhinoDale, 2020-11-20:
Thank you I will try/correct this

------------------------------
Dale Bratcher,
Senior System Administrator and Lead Software Engineer for
K-Kel Inc, DBA Spearmint Rhino Adult Cabaret, Las Vegas
Vegas Party Linq, LLC
3340 S HIGHLAND DR
LAS VEGAS NV 89109-3427
dale@spearmintrhinolv.com
Direct: 702-797-0494

closed 2020-12-10 by todbot