node-hid/node-hid #406: Device Not Detected

farizyoga, 2020-11-22:
Hello, i have Embedded 2d scan module Taffware EP3000, but the device is not recognized when i call HID.devices()

im running on macos

how to fix this?

todbot, 2020-11-22:
It appears that the EP3000 barcode scanner has multiple USB modes.  in USB HID working mode, it acts as a keyboard, and the OS owns keyboards (for security reasons).  `node-hid` cannot access keyboard devices for this reason.  

However it appears the EP3000 can be set to USB serial port working mode.  In this mode, it appears as a serial port devices to the OS.  For that you would use `node-serialport` not `node-hid`.

closed 2020-11-22 by todbot

farizyoga, 2020-11-24:
Thank you @todbot 

im able to detect the device using node-serialport