node-hid/node-hid #407: Does not write to the remote HID device.

antt0n, 2020-11-27:
Hello,
The device in question is an aRGB HUB.
I know the buffer is good, because it worked on my side in C#.
Do you have an idea?

`var hid = require("node-hid");

// VID : 1c57- 7255
// PID : 7ed0 - 32464

var device = new hid.HID(0x1c57,0x7ed0);

var buffer = Buffer.alloc(64);
buffer[0] = 57;
buffer[1] = 0,
buffer[2] = 0;

device.write(buffer);`

todbot, 2020-11-27:
Try making `buffer[0]` be zero and shifting the rest of your data down.  The first byte of the buffer is the reportId.

todbot, 2020-11-27:
Also, do you have a link / model number for the "aRGB HUB" you are using? I like to get as many USB HID devices as possible to test against.  Thanks.

antt0n, 2020-11-27:
Thank you for your prompt response!
I have already tried the solution you propose.
I have tried absolutely everything possible.
For the moment, I really think it's a problem with the module.
Knowing that when I start writing, the console doesn't tell me anything, it's like it's running in a vacuum.
The HUB in question is the Zalman Z-Sync.

todbot, 2020-11-27:
Your buffer must be exactly the size of the report, plus one for the reportId if they are used. Do you have the USB HID Report Descriptor for this device?  That describes the type and sizes of reports sent or received by the device. In C#, Windows takes care of some of that for you, but we don’t have those conveniences in Node.

antt0n, 2020-11-27:
Problem solved!
To answer you, there is no data sheet or specifications for this HUB, I do everything by reverse engineering.
I actually solved my problem by following your instructions, which gives me this code:

```
var hid = require("node-hid");

// VID : 1c57- 7255
// PID : 7ed0 - 32464

var device = new hid.HID(0x1c57,0x7ed0);

var buffer = Buffer.alloc(66);
buffer[0] = 0;
buffer[1] = 57;
buffer[2] = 1,
buffer[3] = 100;

device.write(buffer);
```

Thank you very much!

closed 2020-11-27 by antt0n