node-hid/node-hid #408: Document UV_THREADPOOL_SIZE env var

Richie765, 2020-11-28:
I ran into a nasty problem when watching multiple keyboards using this module. To make a long story short, if you want to watch more than a few HID devices, you have to set UV_THREADPOOL_SIZE to a higher value (the default is 4). Otherwise the event-loop blocks resulting in unpredictable results that are hard to debug.

todbot, 2020-11-28:
Interesting. This seems more like a general Node I/O concern than specifically about `node-hid`.  How many devices are you using concurrently when you see this problem occur?  Can you provide example code so I can try to replicate the blocking? Thanks!

todbot, 2020-11-28:
Also, how are you watching keyboards?  Normally `node-hid` does not have access to USB HID keyboards because the OS owns them.

Richie765, 2020-12-04:
@todbot, it is a general Node.js concert, but not quite. In general situations you would never encounter this issue because most I/O will give a result in a short time. Not with HID however, as it will block until input is received. Since by default there are 4 threads, you can monitor 4 HID devices without issue. However, you cannot do any FS I/O at the same time as they will also eat up a threat (network is generally fine, sync FS I/O _may_ be fine too). The rights on the /dev/hidraw* are set so they can be accessed, no problem there. Here's an example:

```javascript
const HID = require('node-hid');

function monitorDevice(path, name) {
  const device = new HID.HID(path);

  console.log(`${ path }: Start monitoring`);

  device.on('data', data => {
    console.log(`${ name } =>`, data);
  });

  device.on('error', err => {
    console.error(`${ name } =>`, err);
  });
}

monitorDevice('/dev/hidraw0', 'A');
monitorDevice('/dev/hidraw2', 'B');
monitorDevice('/dev/hidraw4', 'C');
monitorDevice('/dev/hidraw6', 'D');
monitorDevice('/dev/hidraw8', 'E');
```

So first comment a couple of them to be sure they are all working. Then try 4 of them, still fine. With 5 weird things start to happen, e.g. keypresses not comming through, then they come through later on when you press a key on a different device, all jumbled up. When you do FS I/O elsewhere or monitor more HID devices, it gets more confusing.

Julusian, 2020-12-11:
I have noticed this before too, I think it was the read function which [runs in a uv worker](https://github.com/node-hid/node-hid/blob/32d92b5e47306ed3f0cdd9990a3dcd54cf249c5a/src/HID.cc#L189) and it runs a loop which hogs the thread until there is data to return.

This will have performance implications for the rest of the node app if running even just the one read, as it will slow down other io operations as less will be able to be executed in parallel.

I guess the solution would be for node-hid to spin up its own threads. Perhaps one per device upon creation, to be used for all io operations for that device.

neppi, 2021-05-20:
@Julusian I think we have the same problem having more then 4 HID Devices. With one thread per device do you mean node-hid should be implemented using real threads? Or do you mean my node application should deal with problem?

We can contribute the information having this problem on raspbian, windows and ubuntu.

Has anybody solved this issue?

Julusian, 2021-05-24:
@neppi I think that the proper solution is for node-hid to use real threads instead of the libuv pool.

But until that is done, you can run your hid devices inside worker_threads. It adds some complication, but I believe it resolves the issue if you split them with up to 4 devices per worker_thread

mcuee, 2023-05-06:
#451 has been merged. Maybe this issue can be closed?

closed 2025-06-03 by Julusian