node-hid/node-hid #409: Execute the write in asyncronous.

antt0n, 2020-11-29:
Hello,
In case the HID device softbricks, it is always detected by Windows, but does not respond to requests that might be sent to it.
Is it possible to stop a write attempt in case the device does not respond after a certain number of seconds?
Thank you in advance.

pierreminiggio, 2020-11-29:
Same issue here.

RedsTom, 2020-11-29:
Same for me

todbot, 2020-11-29:
So you're saying that `device.write()` blocks on Windows?

antt0n, 2020-11-29:
If you mean by "blocks" that it causes the current JS program to freeze, yes, that's right.
But, it happens only when I put my HID device in a special state: it is recognized by Windows, but does not respond to requests because it is brick.

todbot, 2020-11-29:
Does the blocking also happen in other OSes like Linux or MacOS?
Do you have example code you could show me that demonstrates the problem?  
(I understand I don't have the same hardware as you, but having a small test case would be useful to see and I could perhaps create a similar hardware device that behaves the same way.)

antt0n, 2020-11-29:
I didn't try under other OS, because I plan to use node-hid in an application context under Windows (I can try anyway).
I can give you a relatively simple example with which I encounter the problem:
The device with which I encounter this problem is the Zalman Z-Sync (it is an aRGB hub). The software delivered with this product is natively not very compatible with Windows, and can cause the brick of the aRGB hub.
```
let hid = require("node-hid");
let device = new hid.HID(0x1c57,0x7ed0);

let buffer = Buffer.alloc(66);
buffer[0] = 0;
buffer[1] = 57;
buffer[2] = 1; // Channel
buffer[3] = 155; // Luminosité

device.write(buffer)
```

todbot, 2020-12-10:
Hi,
`node-hid@2.1.1` has been released that uses an updated version of `hidapi`. Please try your code again with the updated `node-hid` and see if you experience the same issues.

closed 2025-06-03 by Julusian