node-hid/node-hid #410: Feature request: asynchronous API

doug-a-brunner, 2020-12-02:
Some operations can take a long time to return, especially on Windows, thanks to its wonky USB stack. Getting a device list seems to be the worst offender; it usually completes in 10-20 ms on my machine, but sometimes takes > 5000 ms for no reason I've been able to discern. I/O to devices can also hang up if the peripheral is having trouble. Right now I have helper processes performing all these calls, to avoid blocking the UI (encapsulated in [node-hid-async](https://github.com/hgmelectronics/node-hid-async)). This seems a bit clunky, though.

Now that everything is ported to N-API, is it easier to provide an async API for these potentially long-running operations?

todbot, 2020-12-10:
Hi @doug-a-brunner,

This is a really interesting addition. At first glance it seems more complex than I'm able to maintain. Just keeping `node-hid` working across all the different OSes and Node and Electron versions has been a real chore. N-API helps, but I'm unclear how much extra cognitive load the addition of Typescript and RXJS will cause. (I know nothing of them beyond occasionally seeing Typescript and spending 30 minutes now reading about RXJS)

I guess my questions would be:
1. Can we make these APIs optional, to not break existing code?
2. Can users still use normal Javascript (not Typescript) when using these APIs?
3. Does RXJS have any negative impacts when used inside of Electron or with `electron-remote` IPC?
4. Would it make sense to have `node-hid-async` as a separate package and then under the `node-hid` org? (if you'd like to not be a long-term maintainer of it)

Also, I just released `node-hid@2.1.1` that includes an updated `hidapi`.  Perhaps it solves some of your `HID.devices()` blocking.  USB enumeration is expensive in general, especially if the USB hub chain is long, so you may not seen much improvement.  What is the topology of your USB setup? (i.e. what does something like [USBTreeView](https://www.uwe-sieber.de/usbtreeview_e.html) show as your deepest USB "leaf"? I think Windows does a "better" job of re-interrogating the entire USB space while MacOS/Linux does a bunch of caching, and that could be why it takes longer. I don't think there's a bug in hidapi's enumerate but if there is, we should investigate that.

Julusian, 2020-12-11:
@todbot I think the main thing to tackle here is to have an async version of the potentially slow c++ functions. Like how there is read and readSync.
Having some alternate functions which run in a background thread (or uv worker) and fire a callback when they complete is all I think is necessary. This means that other js code will be able to execute while the usb devices enumerate, which right now is not possible without doing the usb work in a child process like node-hid-async is doing.

Thread safety is a potential concern here, but it will be managable.

As for your questions:
1) Yes, we can either detect the presence of a callback function when the method is called, or have a second function with a different name
2) Yes, before being published to npm it gets compiled to js and some type definitions. So a typescript library can be used by a pure js project just fine.
3) I dont know much about rxjs so I cant give a good answer here, but I don't think it is really necessary, personally I would go with a simple EventEmitter. 
4) Possibly, if every potentially slow api method in this library has a async/callback based version, then that will be enough to make node-hid-async not necessary, but simply a different opinion on what the library api should look like.

Julusian, 2025-06-03:
v3.0.0 introduced a proper async api

closed 2025-06-03 by Julusian