node-hid/node-hid #412: Electron 10 on Windows crashes with EXCEPTION_ACCESS_VIOLATION_READ

rmtry, 2020-12-18:
Hi,
I have an Electron project to read/write USB devices using node-hid@2.1.1 and usb-detection@4.10.0
It happens to be fine on macOS, but on Windows it crashes a lot. I tried to round up the problem thinking this is an Electron issue but it appeared to be fine without USB read/writing action so I supposed it belongs to here.

I tried to look for the solution online but I was unable to find a related one. Has anyone witnessed this before?

[image]

todbot, 2020-12-18:
Hi @rmtry,
Can you post a code sample that replicates what you're seeing?

One issue I have seen when using `usb-detection` and `node-hid` together is that there is a time delay in the OS between when the USB device is first detected and when it's actually been installed in the OS's HID stack.  This delay can be several seconds.

So the first thing I would try is to put a 5 second delay before doing anything with `node-hid` after triggering on a `usb-detection` event.

rmtry, 2021-01-18:
Hi @todbot,

We had found the problem is that too many HID were generated. For every USB action, we created a new HID connection and close after. I don't know why on macOS it was fine and Windows crashed but not throwing any specific error.
Generating only 1 connection util the device is disconnected fixed the problem.

And indeed it takes a while before the HID recognize the device found by usb-detection. This happens to solve another issue  of our project. Thank you very much!

closed 2021-01-18 by rmtry