node-hid/node-hid #413: [Raspberry Pi] Incorrect Result from barcode scanner

uzumakinaruto123, 2020-12-23:
Trying to scan bar codes in electron app, scanning actual bar code with value **8901042960340** 

Already allowed permission using udev and also getting the result using device.read()

`
scannerDevice.read((err, data) => {
        console.log(data);
    });
`

**On scanning output is as below:**
`
Uint8Array(8) [0, 0, 37, 0, 0, 0, 0, 0]
`<br>
With all bar codes, the result is similar with all zeros except 1 entry with number.

<br>

**Expected Output:**
 
`8901042960340`

**OR**

`
Uint8Array(13) [56, 57, 48, 49, 48, 52, 50, 57, 54, 48, 51, 52, 48]
`

<br><br>

On windows, tried the browser based onscan.js, it returns the correct result. 

Am I missing something?

todbot, 2020-12-23:
Is this scanner a keyboard-acting device?  If so, you will have inconsistent and mostly invalid results until you unload the HID keyboard driver.

Does your code send any initialization to the scanner? (any `device.writes()` or `device.sendFeatureReports()`)

uzumakinaruto123, 2020-12-24:
I am new to this barcode scanner research so am not sure. It's a 1d bar code scanner and mostly keyboard-acting device.

If this helps: [Product link](https://www.amazon.in/Tvs-BSC-101-STAR-Barcode-Scanner/dp/B074W9VXM5/ref=sr_1_3?crid=2NIFDLT9OLZFH&dchild=1&keywords=tvs+barcode+scanner&qid=1608792352&sprefix=tvs+bar%2Caps%2C389&sr=8-3)

I don't have any `.write()` or `.sendFeatureReportCalls()` in the code

>  unload the HID keyboard driver

By this, you mean uninstalling keyboard drivers and reinstalling?

todbot, 2020-12-24:
As [mentioned here](https://github.com/node-hid/node-hid#keyboards-and-mice), `node-hid` cannot access USB HID keyboards for security reasons.  Otherwise you could trivially build a keylogger.  

The `onscan.js` library is using standard Javascript browser keyEvents to read keypresses, as if you were typing on the keyboard.  

There are two ways to have a Node app read from this barcode scanner:
- Set the scanner to RS-232 mode (See the [BS-C101 Star installation / programing guide](https://www.tvs-e.in/bs-c101-star/) for details) and use `node-serialport` instead of `node-hid`.
- Use standard Node keypress reading functions like `process.stdin.on()` to read from the "keyboard" that is the scanner. This is essentially emulating what `onscan.js` does.

If you are writing an Electron app, since it's a browser, you can use `onscan.js`.

closed 2020-12-24 by todbot