node-hid/node-hid #414: No Data Sent From Xbox One Controller

bennyman123abc, 2020-12-29:
Hi there! So I'm having an issue getting my Xbox One controller to communicate with node-hid.

I can get it to show up in HID.devices(), and I can get node-hid to open it, but the controller sends no data. The controller I'm using is a wired Amazon Basics Xbox One controller, with a VID of 0x045E, and a PID of 0x02FF.

The code I've written for this controller consists of:
```js
const HID = require("node-hid");
var devices = HID.devices();

var device = new HID.HID("\\\\?\\hid#vid_045e&pid_02ff&ig_00#7&eb40f38&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}")

console.log(devices);

device.on("data", data => {
    console.log(data);
});
```

The exact same code (Except for the device path changed) works with an official Sony PS4 controller with no issue, and receives data with no issue either.

Does anyone know what I'm doing wrong here?

bennyman123abc, 2020-12-29:
I believe it's also worth noting that clicking between windows sends a data packet of:
`<Buffer 00 80 00 80 00 80 00 80 00 80 00 00 00 00 00>`

todbot, 2020-12-29:
What OS is this on? Looks like maybe Windows? In general, you should not hardcode paths like that because they can change depending on which USB port you plug in to and which HID Usage you are attempting to access, if the device provides multiple usages.

If the controller is reporting on one of the standard HID Usages or you have game controller driver installed, then the OS/driver will claim the device and you may not be able to connect to it meaningful.  If so, you will need to detach the driver in order to use `node-hid` (which is only possible maybe on Linux).  It also may be the controller is only a HID device for some minor aspect (e.g. setting the lights on it or for media buttons) and the actual controller is some other USB protocol.

Otherwise, you can use whatever joystick APIs for your OS instead of `node-hid`.  For instance, on Linux there is the "joyutils" package that includes `jstest`, among others: https://github.com/datrh/joyutils

If your intent is to create an Electron app, you may have better luck with in-browser APIs, such as the Gamepad API: https://developer.mozilla.org/en-US/docs/Web/API/Gamepad_API/Using_the_Gamepad_API

This may be a dupe of #193.

bennyman123abc, 2020-12-29:
Forgot to include the OS, sorry about that. The OS being used is Windows 10. I can try installing an alternative driver and determining if the device still appears in the device list, as if maybe it is being claimed by the OS. I can also look into the possibility of unclaiming it.

bennyman123abc, 2020-12-29:
So, I found, and believe it's worth noting, that inputs are captured if you install generic HID Compliant Game Controller drivers. This may not be ideal for some, but should work fine for my purposes. I'll go ahead and close this issue, and it can be reopened if someone else needs to use the default driver, and developments can be made.

Thank you for your help!

closed 2020-12-29 by bennyman123abc

dii393, 2021-05-26:
Hello guys, my name is diego i'm not very good with english and so i'm using google translator to help me ask this question. I'm trying to get node-hid to communicate with the xbox one controller, but it doesn't send a buffer. Does anyone know of an alternative way to communicate node-hid with xbox-one control. 
My OS is Windows 10.