node-hid/node-hid #416: Cannot write to hid device

T4lus, 2021-01-14:
Hi

I'm trying to write data to an custom HID Device (arduino Atmega32U4, rawhid)
and I'm geting `Uncaught TypeError: Cannot write to hid device` 

here is my code :
`
var os = require('os');
var HID = require('node-hid');

var devices = HID.devices();

console.log(devices);

var isStreamLightController = function(d) { 
    return d.vendorId===0x1209 && d.productId===0x4284;
}

var isRawHidUsage = function(d) { 
    return d.usagePage===0xFFC0 && d.usage===0xC00;
}

var deviceInfo = devices.find( function(d) {
    return isStreamLightController(d) && isRawHidUsage(d);
});

if( deviceInfo ) {
    console.log(deviceInfo);

    var device = new HID.HID( deviceInfo.path );
    console.log(device);

    device.on("data", function(data) {
        console.log(data);
    });

    device.on("error", function(err) {
        console.log(err);
        device.close();
    });
    
    let command = [
        0x02, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00,
        0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
    ];
  
    if( os.platform() == 'win32' ) {
        command.unshift( 0x00 );
    }
    console.log('Sending: ', command)
    device.write(command);
 
    device.close();
}
`

here is the result of `console.log(deviceInfo);`

`
interface: 2
manufacturer: "T4lus"
path: "\\?\hid#vid_1209&pid_4284&mi_02#7&24a42312&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}"
product: "StreamLight Controller"
productId: 17028
release: 256
serialNumber: "FB66DF55421900F5"
usage: 3072
usagePage: 65472
vendorId: 4617
`

Here is a dump of the USB Descriptor 
Information for device StreamLight Controller (VID=0x1209 PID=0x4284):

Connection Information:
------------------------------
Device current bus speed: FullSpeed
Device supports USB 1.1 specification
Device address: 0x0001
Current configuration value: 0x00
Number of open pipes: 0

Device Descriptor:
------------------------------
0x12	bLength
0x01	bDescriptorType
0x0200	bcdUSB
0xEF	bDeviceClass      (Miscellaneous device)
0x02	bDeviceSubClass   
0x01	bDeviceProtocol   
0x40	bMaxPacketSize0   (64 bytes)
0x1209	idVendor
0x4284	idProduct
0x0100	bcdDevice
0x01	iManufacturer   "T4lus"
0x02	iProduct        "StreamLight Controller"
0x03	iSerialNumber   "FB66DF55421900F5"
0x01	bNumConfigurations

Configuration Descriptor:
------------------------------
0x09	bLength
0x02	bDescriptorType
0x0064	wTotalLength   (100 bytes)
0x03	bNumInterfaces
0x01	bConfigurationValue
0x00	iConfiguration
0xA0	bmAttributes   (Bus-powered Device, Remote-Wakeup)
0xFA	bMaxPower      (500 mA)

Interface Association Descriptor:
------------------------------
0x08	bLength
0x0B	bDescriptorType
0x00	bFirstInterface
0x02	bInterfaceCount
0x02	bFunctionClass      (Communication Device Class)
0x02	bFunctionSubClass   (Abstract Control Model - ACM)
0x00	bFunctionProtocol   
0x00	iFunction

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x00	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x02	bInterfaceClass      (Communication Device Class)
0x02	bInterfaceSubClass   (Abstract Control Model - ACM)
0x00	bInterfaceProtocol   
0x00	iInterface

CDC Header Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x00	bDescriptorSubtype
0x0110	bcdCDC

CDC Call Management Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x01	bDescriptorSubtype
0x01	bmCapabilities
0x01	bDataInterface

CDC Abstract Control Management Functional Descriptor:
------------------------------
0x04	bFunctionalLength
0x24	bDescriptorType
0x02	bDescriptorSubtype
0x06	bmCapabilities

CDC Union Functional Descriptor:
------------------------------
0x05	bFunctionalLength
0x24	bDescriptorType
0x06	bDescriptorSubtype
0x00	bControlInterface
0x01	bSubordinateInterface(0)

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x81	bEndpointAddress  (IN endpoint 1)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0010	wMaxPacketSize    (1 x 16 bytes)
0x40	bInterval         (64 frames)

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x01	bInterfaceNumber
0x00	bAlternateSetting
0x02	bNumEndPoints
0x0A	bInterfaceClass      (CDC Data)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x02	bEndpointAddress  (OUT endpoint 2)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x83	bEndpointAddress  (IN endpoint 3)
0x02	bmAttributes      (Transfer: Bulk / Synch: None / Usage: Data)
0x0040	wMaxPacketSize    (64 bytes)
0x00	bInterval         

Interface Descriptor:
------------------------------
0x09	bLength
0x04	bDescriptorType
0x02	bInterfaceNumber
0x00	bAlternateSetting
0x01	bNumEndPoints
0x03	bInterfaceClass      (Human Interface Device Class)
0x00	bInterfaceSubClass   
0x00	bInterfaceProtocol   
0x00	iInterface

HID Descriptor:
------------------------------
0x09	bLength
0x21	bDescriptorType
0x0101	bcdHID
0x00	bCountryCode
0x01	bNumDescriptors
0x22	bDescriptorType   (Report descriptor)
0x001C	bDescriptorLength

Endpoint Descriptor:
------------------------------
0x07	bLength
0x05	bDescriptorType
0x84	bEndpointAddress  (IN endpoint 4)
0x03	bmAttributes      (Transfer: Interrupt / Synch: None / Usage: Data)
0x0010	wMaxPacketSize    (1 x 16 bytes)
0x01	bInterval         (1 frames)

Microsoft OS Descriptor is not available. Error code: 0x0000001F

String Descriptor Table
--------------------------------
Index  LANGID  String
0x00   0x0000  0x0409 
0x01   0x0409  "T4lus"
0x02   0x0409  "StreamLight Controller"
0x03   0x0409  "FB66DF55421900F5"

------------------------------

Connection path for device: 
Contrôleur hôte universel Intel(R) gamme ICH10 USB - 3A39
Root Hub
StreamLight Controller (VID=0x1209 PID=0x4284) Port: 1

Running on: Windows 10 or greater (Build Version 18363)

Brought to you by TDD v2.15.0, Jun  8 2020, 17:18:07

todbot, 2021-01-15:
Do you have another app running that controls that device?  Or, is there a Windows device driver installed for this device?

On Windows, you must send a buffer the exact size of the HID Report you writing to, so you will need the HID Report Descriptor, which is what describes the HID reports that a devices sends and receives.  (I believe USBlyzer and HHD's Device Monitoring Studio can do this on Windows. I've not found a simple open source way of getting the HID Report Descriptor)

But it kind of looks like you're sending on reportId 2, in which case you should take out that 'unshift'.

T4lus, 2021-01-15:
> Do you have another app running that controls that device? Or, is there a Windows device driver installed for this device?

No other app controls that device, and Yes there is a Windows driver installed : "USB Composite Device"

> On Windows, you must send a buffer the exact size of the HID Report you writing to, so you will need the HID Report Descriptor, which is what describes the HID reports that a devices sends and [...]

no matter what size of data, I just get the error

here is my HID Descriptor :
```
#undef RAWHID_USAGE_PAGE
#define RAWHID_USAGE_PAGE 0xFFC0
#undef RAWHID_USAGE
#define RAWHID_USAGE 0x0C00

#define RAWHID_TX_SIZE 16
#define RAWHID_RX_SIZE 64

static const uint8_t  _hidReportDescriptorRawHID[] PROGMEM = {
	/*    RAW HID */
    0x06, lowByte(RAWHID_USAGE_PAGE), highByte(RAWHID_USAGE_PAGE),      /* 30 */
    0x0A, lowByte(RAWHID_USAGE), highByte(RAWHID_USAGE),

    0xA1, 0x01,                  /* Collection 0x01 */
    0x75, 0x08,                  /* report size = 8 bits */
    0x15, 0x00,                  /* logical minimum = 0 */
    0x26, 0xFF, 0x00,            /* logical maximum = 255 */

    0x95, RAWHID_TX_SIZE,        /* report count TX */
    0x09, 0x01,                  /* usage */
    0x81, 0x02,                  /* Input (array) */

    0x95, RAWHID_RX_SIZE,        /* report count RX */
    0x09, 0x02,                  /* usage */
    0x91, 0x02,                  /* Output (array) */
    0xC0                         /* end collection */
};
```

> But it kind of looks like you're sending on reportId 2, in which case you should take out that 'unshift'.
I'm sending on ReportId 0
`Sending:  [0,2,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0]`

todbot, 2021-01-15:
I think your device is not configured correctly. I have verified that your Node app works on MacOS and Windows using an ATmega32U4 running Arduino and NicoHood's HID-Project RawHID library.  Details here: https://github.com/todbot/node-hid-tests/tree/main/node-hid-tst-416

closed 2021-01-15 by todbot