node-hid/node-hid #417: Empty path when connecting USB device at end of USB chain

mountaindude, 2021-01-22:
I noted the [still open issue](https://github.com/signal11/hidapi/issues/301) related to long USB chains, I suspect my issue might be related.

What happens is basically that node-hid *sometimes* fails to properly detect devices at the end of long USB chains. 
Using a Macbook Pro connected to a Belkin docking station, to which a couple of USB hubs are attached.
A device attached to either of those leaf hubs end up either not getting a path value, or not being detected at all.

Weird thing is that sometimes things work and node-hid properly detects even those leaf-most devices. If anything it seems to be related to having recently rebooted the computer... So there might be something else at play too.
If the USB device (an X-keys panel) is plugged in directly to the laptop or into the Belkin docking station, node-hid will correctly detect the X-keys  every time (and get its path).

Been testing node-usb-detection too. 
It seems to correctly/reliably identify newly connected USB devices, even when connected at the end of the USB tree.

So..
Is there a workaround for node-hid/hidapi not playing nice with long USB chains?
Or can node-usb-detection be used to detect the USB device, and then pass that info on to node-hid...? Can't see how, but still putting forward the thought.

todbot, 2021-01-22:
Hi @mountaindude 
Can you try downloading [`hidapitester`](https://github.com/todbot/hidapitester/releases/tag/0.2) and doing `hidapitester --list` or `hidapitester --list-detail`.  This basically does the same thing as `HID.devices()` in `node-hid` but using direct `hidapi` calls. This can let us know if the problem is in `hidapi` or `node-hid`.

Also can you attach a screenshot of what "System Information" (option-click on Apple menu) shows for the "Hardware/USB" section so I can get an idea of the topology of your USB setup?

mountaindude, 2021-01-22:
Yes of course, will do. 
I'm new to hidapi and node-hid, so those pointers to tools are very welcome. 
Will try to get some examples of successful and failing enumerations.

mountaindude, 2021-01-22:
Ok, here is a failing setup, with the X-keys XK-80 keyboard connected to a powered USB hub, which is connected to the Belkin docking station, which connects to the Macbook Pro.

[image]

The details for the K-keys panel are:

```bash
➜  Downloads ./hidapitester --list-detail

05F3/0461: P. I. Engineering - XK-80 HID
  vendorId:      0x05F3
  productId:     0x0461
  usagePage:     0x0001
  usage:         0x0002
  serial_number:
  interface:     2
  path:

05F3/0461: P. I. Engineering - XK-80 HID
  vendorId:      0x05F3
  productId:     0x0461
  usagePage:     0x0001
  usage:         0x0006
  serial_number:
  interface:     1
  path:

05F3/0461: P. I. Engineering - XK-80 HID
  vendorId:      0x05F3
  productId:     0x0461
  usagePage:     0x000C
  usage:         0x0001
  serial_number:
  interface:     0
  path:

➜  Downloads

```

mountaindude, 2021-01-22:
And here's a a working setup, with the X-keys panel connected directly into the Belkin dock (which is still connected to the Macbook Pro). 
The `path` field is populated here.

[image]

```bash
➜  Downloads ./hidapitester --list-detail

05F3/0461: P. I. Engineering - XK-80 HID
  vendorId:      0x05F3
  productId:     0x0461
  usagePage:     0x000C
  usage:         0x0001
  serial_number:
  interface:     0
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG1@1,1/IOPP/UPSB@0/IOPP/DSB1@1/IOPP/UPS0@0/IOPP/pci-bridge@2/IOPP/pci8086,15f0@0/AppleUSBXHCITR@00000000/AppleUSB20XHCIPort@00200000/USB2.1 Hub@00200000/AppleUSB20Hub@00200000/AppleUSB20HubPort@00210000/USB2.1 Hub@00210000/AppleUSB20Hub@00210000/AppleUSB20HubPort@00213000/XK-80 HID@00213000/XK-80 HID@0/AppleUserUSBHostHIDDevice

05F3/0461: P. I. Engineering - XK-80 HID
  vendorId:      0x05F3
  productId:     0x0461
  usagePage:     0x0001
  usage:         0x0002
  serial_number:
  interface:     2
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG1@1,1/IOPP/UPSB@0/IOPP/DSB1@1/IOPP/UPS0@0/IOPP/pci-bridge@2/IOPP/pci8086,15f0@0/AppleUSBXHCITR@00000000/AppleUSB20XHCIPort@00200000/USB2.1 Hub@00200000/AppleUSB20Hub@00200000/AppleUSB20HubPort@00210000/USB2.1 Hub@00210000/AppleUSB20Hub@00210000/AppleUSB20HubPort@00213000/XK-80 HID@00213000/XK-80 HID@2/AppleUserUSBHostHIDDevice

05F3/0461: P. I. Engineering - XK-80 HID
  vendorId:      0x05F3
  productId:     0x0461
  usagePage:     0x0001
  usage:         0x0006
  serial_number:
  interface:     1
  path: IOService:/AppleACPIPlatformExpert/PCI0@0/AppleACPIPCI/PEG1@1,1/IOPP/UPSB@0/IOPP/DSB1@1/IOPP/UPS0@0/IOPP/pci-bridge@2/IOPP/pci8086,15f0@0/AppleUSBXHCITR@00000000/AppleUSB20XHCIPort@00200000/USB2.1 Hub@00200000/AppleUSB20Hub@00200000/AppleUSB20HubPort@00210000/USB2.1 Hub@00210000/AppleUSB20Hub@00210000/AppleUSB20HubPort@00213000/XK-80 HID@00213000/XK-80 HID@1/AppleUserUSBHostHIDDevice

➜  Downloads
```

todbot, 2021-01-22:
Interesting, okay this looks like an issue with `hidapi`. I'll look into it on that side of things first.  I think the issue is around [Line 464 of hidapi/mac/hid.c](https://github.com/libusb/hidapi/blob/d15d594a415e5f51d2cd8ff8bb77b3765a539b98/mac/hid.c#L464), since it appears `hidapitester` is getting the empty string for a path.

mountaindude, 2021-01-23:
I thought of something overnight here. 

Maybe that USB hub (via which no paths are assigned) is broken somehow? 
I am using it for other USB devices so it's not totally broken, but it's not the newest one either. 
I'll see if I can hook up a few more (powered) usb hub in series to create a long(er) chain without using that particular hub. 

Just so we avoid spending debugging time due to some hardware issue.

todbot, 2021-01-23:
I don't think it is a misbehaving USB hub. I can replicate your results with a chain of USB hubs.  The problem stems from an underlying Mac system call having a limit of 512 characters and the paths being longer than 512 characters.  There's a potential fix that we're looking into now over at hidapi.

todbot, 2021-01-23:
But for now, a workaround is to plug your device "further up" the USB tree (i.e. have fewer hubs in between the device and the computer).  Note that 8-port USB hubs are implemented as two 4-port hubs, so you can see situations where plugging a device into port 1 works and port 8 fails.

mcuee, 2023-05-04:
The issue has been fixed in upstream HIDAPI project. So this can be closed as well.

mig8447, 2023-10-25:
PR is there, it is only a matter for maintainers to accept it https://github.com/node-hid/node-hid/pull/513