node-hid/node-hid #418: tmp.node is not a valid Win32 application

xiajingren, 2021-01-25:
I use node-hid in electron, and I use electron-builder to package applications. When I packaged the app, I got an HTTP 404 error
[image]
[image]

The software runs well in Windows 64 bit system

But when running in win7 32bit system, an error occurs
[image]

Looking forward to your reply, thank you very much!

todbot, 2021-01-25:
Hi,
This is a bug in `electron-rebuild` (https://github.com/electron/electron-rebuild/issues/520 )  which is used by `electron-bulder`or you may have a `rebuild` script in your `package.json`.

The only work-around I know of is to not use `electron-rebuild`, since the N-API binaries of `node-hid` work now in both Node and Electron. 

If you must use `electron-rebuild` for other packages, change your script to only rebuild those packages, not `node-hid`

xiajingren, 2021-01-26:
Hi. @todbot . Thank you very much for your reply!
This is my package.json :

```
{
  "name": "xxx",
  "version": "0.7.8",
  "private": true,
  "author": "xxx",
  "description": "xxxxxx",
  "scripts": {
    "serve": "vue-cli-service serve",
    "build": "vue-cli-service build",
    "lint": "vue-cli-service lint",
    "electron:build": "vue-cli-service electron:build",
    "electron:serve": "vue-cli-service electron:serve",
    "postinstall": "electron-builder install-app-deps",
    "postuninstall": "electron-builder install-app-deps"
  },
  "main": "background.js",
  "dependencies": {
    "axios": "^0.21.0",
    "child_process": "^1.0.2",
    "core-js": "^3.6.5",
    "driver.js": "^0.9.8",
    "electron-updater": "^4.3.5",
    "element-ui": "^2.13.2",
    "fs-extra": "^9.0.1",
    "lodash-id": "^0.14.0",
    "lowdb": "^1.0.0",
    "node-hid": "^1.3.1",
    "usb-detection": "^4.10.0",
    "vue": "^2.6.11",
    "vue-context-menu": "^2.0.6",
    "vue-i18n": "^8.21.1",
    "vue-router": "^3.2.0",
    "vuex": "^3.4.0"
  },
  "devDependencies": {
    "@vue/cli-plugin-babel": "~4.5.0",
    "@vue/cli-plugin-eslint": "~4.5.0",
    "@vue/cli-plugin-router": "~4.5.0",
    "@vue/cli-plugin-vuex": "~4.5.0",
    "@vue/cli-service": "~4.5.0",
    "@vue/eslint-config-prettier": "^6.0.0",
    "babel-eslint": "^10.1.0",
    "electron": "^9.0.0",
    "electron-devtools-installer": "^3.1.0",
    "eslint": "^6.7.2",
    "eslint-plugin-prettier": "^3.1.3",
    "eslint-plugin-vue": "^6.2.2",
    "prettier": "^1.19.1",
    "sass": "^1.26.11",
    "sass-loader": "^10.0.2",
    "vue-cli-plugin-electron-builder": "~2.0.0-rc.4",
    "vue-template-compiler": "^2.6.11"
  }
}
```

`electron-rebuild` seems to be automatic
I don't know how to control it,I'm just run `npm run electron:build`
how to control it from not rebuild `node-hid`?

Thank you again!

xiajingren, 2021-01-26:
Hi. @todbot .I'm sorry to trouble you again.
i try add `npmRebuild: false` to skip rebuild.
[image]
then all the packages were skipped. how can I just skip `node-hid`?

looking forward to your reply, thank you very much!

todbot, 2021-01-26:
Hi @xiajingren,  Apologies I see you are using an old version of `node-hid`.  The current version is 2.1.1.  I was mistaken in my previous comment.

The problem you are seeing is because there is no prebuilt binary for `node-hid` for Windows 32-bit.  You will need to install Windows compiler tools so `electron-rebuild` can make it for you.  See: https://github.com/node-hid/node-hid#compiling-from-source

xiajingren, 2021-01-27:
Hi @todbot , I have installed the `windows-build-tools` and `python27`
my program works well on Windows 64-bit,only run in the Windows 32-bit system to report an error

` prebuild-install http 404 https://github.com/node-hid/node-hid/releases/download/v1.3.1/node-hid-v1.3.1-electron-v80-win32-ia32.tar.gz`

xiajingren, 2021-01-27:
Hi @todbot .
I upgraded `node-hid` to version 2.1.1,I got two 404 errors
[image]

like this: [install-app-deps fails for node-hid with 404](https://github.com/electron-userland/electron-builder/issues/5581)

todbot, 2021-01-27:
The `prebuild-install` process cannot find a prebuilt binary then it will normally attempt to compile one. Those warnings you are highlighting are just warnings. After that it will attempt to compile.  If it cannot, then it errors out. 

As the output you quote states, please try rerunning the command with `DEBUG=electron-builder` environment variable set to see what the actual issue is.

xiajingren, 2021-01-27:
Hi. @todbot .i'm very sorry to ask you again...

I tried again all day, but the problem is still unsolved(tmp.node is not a valid Win32 application). Now I upgrade `node-hid` to version 2.1.1,then I run `set DEBUG=electron-builder` on the command line,but I don't seem to see any valuable information.My program still reports errors in Windows 32-bit system.

you say: 'The only work-around I know of is to not use electron-rebuild ......'  What should I do？I'm sorry my nodejs level is so bad...

thanks.

xiajingren, 2021-01-28:
Hi @todbot .
Now I have finally solved the problem.

First, compile `node-hid` for `ia32`:
`electron-rebuild -v=9.4.2 -m=D:\WorkSpace\Source\Repos\xxxx\node_modules\node-hid -a=ia32`

And then I put the `\build\Release\HID.node` package and copy to my `\npm-cache\_prebuilds` directory and rename it:

[image]

Then I use `electron-builder` to package my program.
Now my program is working in 64-bit and 32-bit systems.

@todbot . Thank you for your help. Thank you for creating such an excellent `node-hid` library!

closed 2021-09-09 by todbot