node-hid/node-hid #419: Is there a field uniquely linking a device in node-hid with one in node-usb-detection?

mountaindude, 2021-01-27:
I have a scenario where a user might have several identical keypads from some vendor. 

Interacting with those keyboards works just fine, I am now trying to find a solid way of handling disconnect/reconnect scenarios for the devices. 

For example:
1. I have a Node.js app that relies on 1 or more keypads being available via USB. Let's say there are 2 connected keypads, both of identical brand, version, vid/pid etc. Only difference is where in the USB tree they are located, i.e. the path.
2. One of them (keypad1) is removed. I want my Node.js app to be notified of this and mark keypad1 as "disconnected".
3. I can use the node-usb-detection lib to detect removal of the keypad. That lib will give some info on product name, vid/pid etc - but no path. 
4. The question is now: How can I from my Node app determine whether it was keypad1 or keypad2 that was disconnected? There doesn't seem to be any field available in **_both_** node-hid and node-usb-detection that uniquely identifies a device... or?

todbot, 2021-01-27:
Hi @mountaindude,
The best solution is the USB serial number field.  This is a standardized field that many (but not all) devices fill in and is part of the USB Device Descriptor that's available as the `serialNumber` property in the devices returned by `HID.devices()` in `node-hid` and in the `device` returned in the callbacks of `node-usb-detection`.

mountaindude, 2021-01-27:
Bummer, the devices I am working with don't have serial numbers set. 
Might have to brute force this then, basically reconnecting all the keypad instances in the node app, even though just a single keypad has been removed/added. Oh well, should work too.

Thanks for the insight though, much appreciated!

closed 2021-01-27 by mountaindude