node-hid/node-hid #420: Linux prebuilds error on debian 10 & ubuntu 18.04

Julusian, 2021-01-28:
Trying to use v2.1.1 on debian 10 results in an error:
```
Jan 28 11:48:14 companion node[2345]: Error: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /home/julus/companion-src/lib/module/thingm-blink1/node_modules/node-hid/build/Release/HID_hidraw.node)
Jan 28 11:48:14 companion node[2345]:     at Object.Module._extensions..node (internal/modules/cjs/loader.js:1057:18)
Jan 28 11:48:14 companion node[2345]:     at Module.load (internal/modules/cjs/loader.js:863:32)
Jan 28 11:48:14 companion node[2345]:     at Function.Module._load (internal/modules/cjs/loader.js:708:14)
Jan 28 11:48:14 companion node[2345]:     at Module.require (internal/modules/cjs/loader.js:887:19)
Jan 28 11:48:14 companion node[2345]:     at require (internal/modules/cjs/helpers.js:74:18)
Jan 28 11:48:14 companion node[2345]:     at bindings (/home/julus/companion-src/lib/module/thingm-blink1/node_modules/bindings/bindings.js:112:48)
Jan 28 11:48:14 companion node[2345]:     at loadBinding (/home/julus/companion-src/lib/module/thingm-blink1/node_modules/node-hid/nodehid.js:19:46)
Jan 28 11:48:14 companion node[2345]:     at Object.showdevices [as devices] (/home/julus/companion-src/lib/module/thingm-blink1/node_modules/node-hid/nodehid.js:126:5)
Jan 28 11:48:14 companion node[2345]:     at _blink1HIDdevices (/home/julus/companion-src/lib/module/thingm-blink1/node_modules/node-blink1/blink1.js:13:14)
Jan 28 11:48:14 companion node[2345]:     at Function.devices (/home/julus/companion-src/lib/module/thingm-blink1/node_modules/node-blink1/blink1.js:19:3)
```

If I check the versions of glibc available:
```
julus@companion:~$ strings /usr/lib/x86_64-linux-gnu/libstdc++.so.6 | grep GLIBC
GLIBCXX_3.4
GLIBCXX_3.4.1
< every version in between >
GLIBCXX_3.4.25
```
I haven't tried running node-hid on ubuntu 18.04, but the command above gives the same output, so I expect it will also be an issue there.

I can probably work around it by forcing a rebuild of the module, but as debian 10 is the [latest release of debian](https://wiki.debian.org/DebianReleases), I feel like it should work with the prebuilds

The fix will just be to compile the prebuilds on ubuntu 18.04 or debian 10.

ert78gb, 2021-02-28:
Hi @todbot,

We have the same issue with our new release. I investigated what is the issue. I hope I can write clearly.

The root cause is the `node-hid@2.x.x` released with a newer version of c++ lib and the older OS-es do not contain the referenced libraries.
If someone rebuilds the `node-hid` with an earlier c++ version that it is working properly on the newer and older Linuxes too.

I think the solution is 2 very simple solution
- change the value of the `run-on` field to `ubuntu-18.04` instead of the `ubuntu-latest`. 
- set the `CC: clang-9` and `CXX: clang++-9` environment variables.

History:
- Last December node-hid changed to use NAPI - thx for it.
- Last December GitHub started the [rollout](https://github.com/actions/virtual-environments/issues/1816) of the `ubuntu-20.04` that means the `ubuntu-latest` will `ubuntu-20.04` that changes the default `Clang` version from `9` to `10`

todbot, 2021-03-01:
Hi @ert78gb and @Julusian ,
Thank you for the detailed report. It helped me discover the root of the build issue.
As a quick hack update, I re-made and re-uploaded the prebuild `node-hid-v2.1.1-napi-v3-linux-x64`.  I've verified this new prebuild works on Ubuntu 18.04.5 and Ubuntu 16.04.6.

You can try it out by reinstalling `node-hid` after flushing your npm prebuild cache. This can be done from within your project with something like:

```
rm ~/.npm/_prebuilds/*node-hid*  # clean prebuild cache for your user
rm -rf node_modules              # remove any possible old node-hid in your project
npm install node-hid             # reinstall node-hid in your project
```

Please let me know if this works for you.  

_Note for myself why this occurred_: I had inadvertently updated libstdc++ on my main build VM. We're not using Github Actions for generating prebuilds yet. I'm looking into that now. It does seem that setting version to ubuntu 18 keeps glibc at the more compatible level. Clang is not required but it does produce smaller binaries, which is nice, at the expense of a warning about the `-exceptions` flag.

Julusian, 2021-03-01:
Thanks, I can confirm this now works on Debian 10

ert78gb, 2021-03-01:
I also tested and works.
Thanks for the quick response and release.

I think would be good if you create a new release version with the new build setup. It more transparent, there was a change.

todbot, 2021-03-01:
Agreed. I currently build on private VMs that I (normally) keep as constant as possible. But with Github Actions being available on three major platforms (Windows x64, MacOS x64, and Linux x64), I think we could have the actions create prebuilds for those three.

Julusian, 2021-03-01:
Yes, I recommend building with github actions.
I am doing this for some of my native node libraries (eg https://github.com/Julusian/node-jpeg-turbo/blob/master/.github/workflows/prebuild.yaml). That script will suffer from the same issue as this thread, as I havent pinned the ubuntu version yet.

Whenever I push a git tag, it will generate the prebuilds, attach them to the release and then publish the release. Nice and automated :)

todbot, 2021-03-01:
Oh yes we're already using github actions for CI and the badges (https://github.com/node-hid/node-hid/tree/master/.github/workflows) and I'm happy to see the prebuild tests I've been doing in another repo look similar to what you've got.  Nice to know I'm on the right track.  Thanks for providing an existence proof of it prebuild uploading working with Actions.

ert78gb, 2023-01-29:
Hi @todbot ,

This issue came up again. 
One of our user [reported](https://github.com/UltimateHackingKeyboard/agent/issues/1961) `node-hid@2.1.2` can't run on RedHat Linux 8. It is a very long term supported linux. The end of the support will 2029-31.

I just would like to ask what is the minimum required version of standard c lib policy of the `node-hid`?

Julusian, 2023-01-29:
It sounds like this has reappeared. redhat 8 is using 2.28, which is newer than ubuntu 18.04(2.27) and the same as debian 10.

That said, doing an `ldd -v node_modules/node-hid/build/Release/HID.node` doesn't make any mention of a version newer than `GLIBC_2.17`. I don't know if that is accurate, or if there is something else I should do to check?

What does `ldd -v` look like for you?

ert78gb, 2023-01-29:
Hi @rf-triegel,

I am inviting you to help investigate the [UHK Agent 2.1.1 is not working on RHEL 8](https://github.com/UltimateHackingKeyboard/agent/issues/1961) issue, because I don't have RHEL 

Could you answer to Julian's question from the prev comment? 

Thx

Julusian, 2023-01-29:
> Error: /lib64/libstdc++.so.6: version `GLIBCXX_3.4.26'

Ah, so its a GLIBCXX version issue not glibc. From some googling, the two are different..  
It looks like the versions of things here are tied to gcc version instead https://gcc.gnu.org/onlinedocs/libstdc++/manual/abi.html.

So with that in mind, redhat 8 and debian 10 looks to support up to 3.4.25, ubuntu 18 is 3.4.24.  

So in this case, it looks like it was built with a too new version of gcc/g++. I did not know that this could be an issue, so is interesting to learn that it can be. Maintaining support for redhat 8 might start to get hard soon though with ubuntu 18.04 being eol soon, and github actions removing images for it soon too. 
#490 is already automating some prebuilds to be done in docker, and could easily make the linux-x64 version be done in docker too, with an image based on debian 10

rf-triegel, 2023-02-14:
The ABI considerations matter for all dynamic-library dependencies. I don't know what's practical for Node, but compare, for example, Python's manylinux approach to builds: https://github.com/pypa/manylinux

closed 2024-07-25 by Julusian