node-hid/node-hid #421: Being able to read data- sometimes.

NabeelAhmed1721, 2021-02-09:
Hey guys,

I was using `node-hid` to read data coming from a keyboard. Everything works correctly, however, reading data happens sometimes. Sometimes it works right away by typing `sudo node index.js` or sometimes I have to `ctrl + c` and repeat that command several times to get it to work. 

I'm not sure how to investigate further into this issue as I can't find a reliable way to determine `node-hid`'s state when it's trying to read the keyboard info. 

What's the best way to solve this issue besides just re-running to command and hoping it works. (Btw I am on macOS if that's of any use).

todbot, 2021-02-10:
`node-hid` cannot read standard HID keyboards (if it can it's a fluke, see https://github.com/node-hid/node-hid#devices-node-hid-cannot-read )  This is an OS-level restriction, nothing `node-hid` can do.

NabeelAhmed1721, 2021-02-10:
Tanks for reply, and yeah, I experimented and I realized it was that specific keyboard. I plugged in another model and it was working 100% perfect of the time. Oh well!

closed 2021-03-01 by todbot