node-hid/node-hid #422: Properly handle device unplug

sensslen, 2021-02-14:
I am very new to this library and USB handling in general. 
What I would like to acheive is to be able to gracefully handle unplugging of a opened HID instance. 
At the moment I am trying to close once I get the error. Unfortunately this seems to not properly close the device, as when I try to reopen the device after it has been plugged in again, the device is not listed in the array returned by a call to `devices`. 
How would I be able to reconnect to a device that has been unplugged during operation?

todbot, 2021-02-14:
Hi @sensslen,

Are you refetching `HID.devices()` to get a new list of devices?  `node-hid` does not do device insertion/removal detection (you can use `node-usb-detection` for that), but this is the known-used pattern for approximating it with just `node-hid`: Occasionally do `deviceList = HID.devices()` to do a bus enumeration to query what HID devices are currently plugged in.  It's not as good as hooking into the OS-specific device add/removal systems that `node-usb-detection` does, but it can work.  Note the array returned by `HID.devices()` is only valid at the time it is called. That array is not updated as the device list changes.

closed 2021-02-14 by todbot

sensslen, 2021-02-15:
Hi @todbot,

thanks a lot for your response. I am actually using `node-usb-detection` to detect insertion of a usb device. I assume my problem comes from another direction. I am using the EventHandler interface to register for data updates from the USB device. When I unplug the device I do get an error - as expected. When I do plug in the same device again, the device gets picked up correctly by `node-usb-detection`. Since I am trying to pick the correct device by filtering for serial Number I am using `HID.devices()` to get the device list following it's detection to iterate the list and find the correct path to connect to.

Unfortunately the call to `HID.devices()` does not pick up the device that I konw was plugged in because I get a notification via `node-usb-detection`. How would I handle this scenario correctly?

For reference: please find my use case in https://github.com/sensslen/node-gamepad (be sure to switch to development branch).

todbot, 2021-02-15:
Hi @sensslen,
I think the issue you're running into is the rather large time delay between when a device is inserted (and sending a `node-usb-detection` event) and when the OS has it hooked into its HID subsystem (thus viewable in `HID.devices()`)  This time delay can be several seconds.  

As a work-around to start, you might try doing something like a `setTimeout( { ... connectIfMatching(...)}, 5000)` in your code.  Alternatively, just do a delay loop printout out the contents of `HID.devices()` on the `usb-node-detection` "add" event to see how long the delay is for you.  

Note that this delay is not constant, and depends on system load and OS (I think Windows takes much longer because it hooks HID devices up to more software stacks internally)

sensslen, 2021-02-16:
Hi @todbot,

thanks again for your quick response. I just implemented a test for your suggestion. I assume that the issue has to do with my filtering. I am currently trying to leverage a technique where I can filter HID devices by serial number. Unfortunately I did not find a way to create a HID instance by serial number. That's the reason I am using ''HID.devices'' to find the correct path. Just now I found that the serial number field is not set in the Array returned. I know the device has a serial number, as it is returned by ''usb-node-detection''.

todbot, 2021-02-16:
Hmm. What OS are you on, what is the device you're inspecting, and what does the serial number look like?  The `serialNumber` property is definitely a returned value in `HID.devices()` and is known to work.

sensslen, 2021-02-16:
I am currently using Windows. This is what I get from `usb-node-detection`: `{"locationId":0,"vendorId":1133,"productId":49730,"deviceName":"Xbox` 360-Controller für Windows","manufacturer":"Microsoft","serialNumber":"9A50CB41","deviceAddress":23}`

HID.devices returns `[{"vendorId":1133,"productId":49730,"path":"\\\\?\\hid#vid_046d&pid_c242&ig_00#9&302c7fb8&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","product":"Controller (XUSB Gamepad)","release":0,"interface":-1,"usagePage":1,"usage":5},{"vendorId":1133,"productId":49730,"path":"\\\\?\\hid#vid_8087&pid_0ac2#6&e65bd83&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}","release":0,"interface":-1}]` when I query `HID.devices` with the same vendorId and productId...

todbot, 2021-02-16:
Ahh, okay.  So the Xbox 360 controller is not an actual HID device.  If you plug it into a non-Windows computer (or use a USB bus analyzer on Windows), you'll see that the controller presents four vendor-specific interfaces, none of which are HID.  On your Windows box, you have installed an "XUSB Gamepad" driver that is creating a "fake" HID device on top of the controller, so it can be used in games like a standard HID gamepad.

What `node-hid` is seeing is this "fake" HID device created by the driver.  And this driver isn't propagating the serial number property of the device.

sensslen, 2021-02-16:
Oh, I see - sorry for bothering with this very special use case. In that case I will just use any other kind of controller :) Thanks a lot for your help!

Regards, Simon

todbot, 2021-02-16:
Haha not a problem. HID is full of pitfalls like this. I'm sure you'll come across another weird situation before too long. Good luck with the `node-gamepad` project!