node-hid/node-hid #423: Cannot open device when used in compiled application

MadlyFX, 2021-03-19:
I'm not sure if this is due to the API switch, but I can't seem to get Node-Hid to work in either an electron subprocess or a standalone executable built with pkg, nexe etc.

As far as I can tell, the HID.node package isn't being recursively added by those frameworks--so when it's called, I'm just getting an "Cannot open device with vendor ID XXX" error.

Adding the node-hid.node and HID.node files next to the packaged exe doesn't help either.

todbot, 2021-03-19:
Hi,
You'll need to be more explicit what your setup is: 
- OS type & version
- Node/Electron version
- how you're packaging `node-hide` (are you using `electron-builder`, etc)

Additionally It sounds like you're not getting very good logging from your application. If the shared library (.node) cannot be found, Node/Electron will error out with a very obvious error message printed to stderr.

Can you create a simplified example showing the issue you're having?  E.g. something like  https://github.com/todbot/electron-hid-test

MadlyFX, 2021-03-19:
Thanks, looks like this was a classic PEBCAK error--I was just passing the serial number of the device to open, and using the wrong process.argv index to get it.

Apologies! Thanks for the great library :)

closed 2021-03-19 by MadlyFX

todbot, 2021-03-19:
haha not a problem. I make those errors all the time! have a good weekend.